MrChromebox/firmware #90: Lenovo N20P black screen after full UEFI firmware install

huj3r, 2018-02-14:
Today I installed the full UEFI ROM on a Lenovo N20P through the firmware utility script and then rebooted; now I always get a black screen (not lit) and it requires a long press on power button to turn off.

Booting with or without a UEFI boot USB stick does not change the result.

Has this model been tested with the UEFI rom? It is listed as supported. The installation script reported it was installing a ROM created on 4 February 2018 if I remember correctly.

Can somebody help me to recover this situation please?

MattDevo, 2018-02-14:
I replied on reddit, need to get my hands on a CLAPPER device to test/diagnose it would seem.

MrChromebox, 2019-08-07:
CLAPPER issues should be resolved in the imminent 4.10 release

closed 2019-08-07 by MrChromebox