MrChromebox/firmware #92: Fan not working in Linux for Google Chromebook Pixel 2013 (Link)

lvdisturbed1, 2018-02-26:
It seems that every other OS I flash on this device kills the fan, causing the device to overheat easily.  Granted, the fan Google provided is kind of crap, but it is a little better than not having one at all.  I have tried Ubuntu, ZorinOS, Kali, PhoenixOS and Bliss OS (both Android x86), as well as Windows 10.  This issue is present in both SeaBIOS and Coreboot.  I have visually verified that the fan operates in ChromeOS and Dev Mode boot screen.

lvdisturbed1, 2018-02-26:
> can you verify that the behavior is the same (fan off) under Linux with
> the stock firmware + RW_LEGACY? Or is that what you mean by with SeaBIOS?

That is what I meant with SeaBIOS.  However, I did select to update the
RW_LEGACY firmware from your Firmware Utility Script.  I will flash the
recovery image to restore stock SeaBIOS when I have time and report.

MattDevo, 2018-02-26:
updating the RW_LEGACY wouldn't affect fan control, so this does sound like a case where the EC expects the OS to perform fan/temp control, vs newer boards where the EC does it automatically (Link was the first device to use Google's ChromeEC, so not surprising).  Perhaps that functionality can be backported

MattDevo, 2018-02-27:
resolved with EC firmware update; will be included in next general firmware update

closed 2018-02-27 by MattDevo

waabamaa, 2018-08-11:
I just installed Ubuntu Studio on Link after using your Firmware Utility Script (latest version) and the RW_Legacy option and am also experiencing this issue. With even moderate use the computer rapidly heats up to the point of being almost too hot to handle. It was functioning fine in ChromeOS immediately prior to this.

waabamaa, 2018-08-15:
ok, thanks. good to know this. i will install the UEFI firmware.

PierceSutton, 2019-10-15:
Hi, I've been having fan issues for a while and have always ignored my chromebook getting very hot. I decided to switch over to your (MrChromebox) UEFI firmware in hopes of solving my heating issues but still no luck. My chromebook is still getting very hot (lm-sensors tells me its +100 degrees, but I don't believe it), any advice? I am using an Acer Chromebook 14 for work, Core i5 (Skylake).

lafrad, 2020-03-22:
Resurrecting an old bug.
Just setup my Chromebook Pixel 2013 (Amazing to have native BIOS!!!!!!) with Ubuntu.
And I'm not hearing the fan run *at all*, and any reasonably heavy processing is feeling really choppy/thermal-throttley.

I've tried Ubuntu 19.10, 20.04 (pre-release), 18.04 (buggy), and a couple versions of GalliumOS.
I know the fan works, because a flash back to stock had it spinning up hard with a couple of youtube tabs open....

What can I do to troubleshoot this?

> ChromeOS Firmware Utility Script [2020-03-17] 
>  (c) Mr Chromebox <mrchromebox@gmail.com> 
> *********************************************************
> **   Device: Google Chromebook Pixel 2013 (LINK)
> ** CPU Type: Intel IvyBridge
> **  Fw Type: Full ROM / UEFI (MrChromebox-4.11.2 03/17/2020)
> **    Fw WP: Disabled
> *********************************************************
>

MrChromebox, 2020-03-23:
just tested here on Mint 19.3 w/kernel 5.3, running `stress -c 4 -i 4` and the fan kicks on in short order

lafrad, 2020-03-24:
Well sure enough.   the stress command gets the fan spinning just fast enough to barely hear it on the 20.04 install. I guess I'll hunt around for what would be causing the system to hang up as much elsewhere?   I was expecting the fan speed to spin up like stock.

MrChromebox, 2020-03-24:
so my EC firmware for LINK makes the following changes: 

* base fan speed raised from 0 to 1200 rpm
* ensure fan is enabled when powered on and disabled when sleeping

But those changes are from 18 months ago.  Additionally, the UEFI firmware sets the fan mode to auto on EC init.

lafrad, 2020-03-24:
Is there any logging I can do from linux?  looks like lm-sensors isn't finding any fans

MrChromebox, 2020-03-24:
no, because it's all handled by the EC