MrChromebox/firmware #94: Massive UEFI issues on TERRA

Wulfsta, 2018-03-12:
I upgraded to the UEFI on an ASUS C301SA and bad things happened. This is intentionally vague, because I'm not really sure what was happening - after leaving the bootloader the machine would often hang at random. Sometimes it would work, sometimes the screen would freak out and the speakers would whine. This happens across distributions - GParted live, GalliumOS, and Arch Linux, are what I tried. I suspect some type of driver issue. Problems nonexistent after reverting to legacy BIOS. Attached are examples of what I'm describing. 
[image]
[image]
I would upload a video of the speakers whining, but github does not support videos.

MattDevo, 2018-03-12:
@reynhout have you seen anything like this on your TERRA?

Wulfsta, 2018-03-12:
I should also mention that I tried reflashing to take into consideration that something may have gone wrong. The problems persisted.

Wulfsta, 2018-03-12:
Updated original post with links to two images and a video of the issue.

reynhout, 2018-03-12:
@MattDevo I have not seen anything like that. I think I flashed the UEFI firmware on TERRA, but only temporarily, and then put it back to factory+RW_LEGACY to test the ChromeOS firmware/Xorg issue.

@Wulfsta How often/quickly does the issue appear? I'll test it tomorrow.

Wulfsta, 2018-03-12:
@reynhout It was happening almost every time I booted. It was more rare for the machine to work than to hang.

reynhout, 2018-03-13:
@Wulfsta I installed the MrChromebox 20180204 full firmware and then reinstalled GalliumOS without running into any issues. Rebooted 5-6 times, everything looks good.

Our TERRA is an ASUS C202SA, hardware ID `TERRA xxx-xxx-xxx-Axx`. The ASUS C300SA and C301SA models  are `TERRA xxx-xxx-xxx-Bxx`, and have different displays so there might well be relevant hardware differences.

Wulfsta, 2018-03-13:
The issues started appearing on stock ChromeOS. I'm chalking this up to unfortunately timed hardware failure and closing the issue. Apologies for any inconvenience caused.

closed 2018-03-13 by Wulfsta