MrChromebox/firmware #97: Enable network in Zako's UEFI firmware

anatol, 2018-05-11:
I am using a Zako for UEFI firmware development. I found that this ChromeOS box is a great and non-expensive way to do development.

I am trying to get `SimpleNetworkProtocol` UEFI protocol that allows to access the underlying network hardware from my UEFI app. My code works in QEMU but at Zako it returns `EFI_NOT_FOUND` error. That I interpret as UEFI firmware does not have support for Zako ethernet chip.

Could you please enable network support for Zako in the UEFI fw?

anatol, 2018-05-14:
UEFI driver from the vendor might help here. See the [vendor download site](http://www.realtek.com/Downloads/downloadsView.aspx?Langid=1&PNid=13&PFid=5&Level=5&Conn=4&DownTypeID=3&GetDown=fa&SortByDesc=1) section "UEFI UNDI Driver".

Could you please release a test firmware for Zako or share some instructions how to modify your existing UEFI firmware and add a new network driver?

MattDevo, 2018-05-14:
I'll have to look into this, as it would require adding both the UEFI networking stack as well as the device-specific driver