MrChromebox/firmware #98: Clarify where to get vendor blobs

anatol, 2018-05-15:
Both `edk2` and `coreboot` build script require vendor blobs (like GOP, ME, FLASHDESCRIPTOR). Without these blobs a user cannot rebuild the image.

Could you please share how one can setup a repo with these blobs? Maybe you can share your repo, or URLs to vendor web sites. Or maybe a script that helps to extract binaries from existing firmware.

MattDevo, 2018-05-15:
these blobs are normally extracted from a stock firmware image, there are generalized instructions on coreboot.org for doing so.  However, some bits are device or platform specific, and things like the GOP driver are extracted from vendor UEFI firmware for a device on the same platform.  It's a big hot mess, and yes at some point I should write up how to do it for devices I support

intika, 2019-10-25:
I confirm, last time i checked those blobs are the same on stock fw ;) 
@MrChromebox  Great, amazing work by the way, thank you for maintaining this

MrChromebox, 2019-10-25:
> I confirm, last time i checked those blobs are the same on stock fw ;)

mostly-ish :) not all platforms/devices use all blobs, but in general:

IFD, ME, MRC, refcode -> from stock FW
FSP, microcode updates -> last Intel public releases
GOP driver -> extracted from other devices' UEFI firmware, or taken from a tool like UBU

intika, 2019-10-27:
Plus : 

EC for older devices (that have it as a blob) -> Extracted from stock

And also there are other firmwares (blob) for other peripherals that are loaded by the Linux Kernel and or from the device flash memory, the list is different for every chromebook (i am referring to some wifi card, touchscreen, tpm, ssd etc.); Note that those blobs element are not located on the firmware provided by @MrChromebox.

We are doomed when it come to trying to use a full open source machine... (we can just try lol)

MrChromebox, 2019-10-27:
> EC for older devices (that have it as a blob) -> Extracted from stock

no, those are all my own custom builds

intika, 2019-10-27:
> > EC for older devices (that have it as a blob) -> Extracted from stock
> 
> no, those are all my own custom builds

Yes i know, i was referring to older devices that does not have an opensource EC like C710 (all devices older than Samsung Chromebook (XE303C12) / Chromebook Pixel 2013).

MrChromebox, 2019-10-27:
none of those use an EC firmware blob as part of the main flash, so they are irrelevant for the purposes of building from my tree

intika, 2019-10-27:
> none of those use an EC firmware blob as part of the main flash, so they are irrelevant for the purposes of building from my tree

Yes indeed :) it's not mandatory to have the ec bloc on the main flash, but i remember that they do get updates over the main flash, when a new version was available.

MrChromebox, 2019-10-27:
> Yes indeed :) it's not mandatory to have the ec blob on the main flash, but i remember that they do get updates over the main flash, when a new version was available.

no, non-ChromeEC devices don't get EC updates that way - they have fixed EC firmware. The update mechanism requires ChromeEC, it's not simply the AP firmware writing a file to the EC's flash. And it's not even built into upsteam coreboot, only my tree has this.

> Anyway... sort of a wiki page with the blobs or something may be useful.

sure, I'll add it at the bottom of my to-do list ;-)

intika, 2019-10-27:
@MrChromebox my bad, you are right, i mix it up a little bit about the ec as i am not actively using those devices... thank you for the reminder :)

bam80, 2022-03-11:
> Anyway... sort of a wiki page with the blobs or something may be useful...

@MrChromebox 
Shall we start maybe, at least with some draft page?

I decided to give it a try and build UEFI fw for my box.
I didn't find an entry point how should I proceed, so had to "reverse engineer" repos here.
For now, I see there are bunch of scripts which are not documented.
Mentioning them in doc would be something to start with.

MrChromebox, 2022-03-11:
with the next release, one will not have to do anything other than clone my repo (and submodules) and then build for their target device. I don't have the time to teach everyone how to build coreboot from scratch, or what blobs are needed for what platforms etc.

closed 2022-03-11 by MrChromebox

mpeter50, 2022-12-11:
> I don't have the time to teach everyone how to build coreboot from scratch,

You wouldn't have to, you can refer to coreboot's docs, or some other source where its already documented.

> with the next release, one will not have to do anything other than clone my repo (and submodules) and then build for their target device.

Is this available, or has it been delayed?

MrChromebox, 2022-12-12:
@mpeter50 it's available now, using the branch from my current release (2022.10.24)