MrChromebox/firmware #99: [Stout] Always-on USB port issue

lis6502, 2018-07-01:
Despite of it's purpose, yellow USB port is not powered after turning off laptop.

lis6502, 2018-07-01:
I believe so- it was described in manual, also i've seen this working on bigger thinkpads, but honestly i did not checked it. 
If there's no one else with stock Chromebook i am willing to revert to stock just to check.

MrChromebox, 2018-07-01:
I don't have anyone else available to test it, so if you're able to that would be helpful so I know if I can use the stock firmware as a reference or not

lis6502, 2018-07-03:
@MrChromebox, according to manual, Always-on USB should be on under these conditions
- Sleep state (breathing i in Thinkpad)
- Shutdown/ hibernation with plugged in charger, but prior to this user should enable this port in Power Manager (Windows OS) 
Hence excuse me, i was convinced that it works OOTB and modified Coreboot just lacks this feature. 
I've reverted to Chrome OS due to firmware change.
Anyways- manual does not mention that ALL USBs are powered while on sleep and none is powered w/wo charged if computer is powered off.

Guess we should ask one who owns Windoze based x131e about this Power Manager features and compare firmware, if applicable (i know that regular x131e runs legacy/ uefi boot mode with regular bios Setup program, so it might be dead-end for Celeron-based Chromebook).

closed 2018-07-03 by lis6502