MrChromebox/firmware #100: Cbfstool utility error msg when flashing script.

Essam113, 2018-07-03:
I got an ASUS C300SA and I followed the instructions step by step, but I'm getting an error msg after executing firmware.sh, the error msg is a cbfstool utility unable to....... etc. Can anyone figure out the reason of this? What did I do wrong?

MrChromebox, 2018-07-03:
just means there was an error downloading a required component, simply try again (bc it's working fine here)

closed 2018-07-03 by MrChromebox

Essam113, 2018-07-03:
Im sorry bothering you... I tried like a 999 times... is there another way to excute the firmware.sh?

MrChromebox, 2018-07-03:
please provide the entire error msg, you clipped the important part

Essam113, 2018-07-03:
Unable to download cbfstool utility
error extracting cbfstool; cannot procced

MrChromebox, 2018-07-04:
this isn't something I can reproduce here.  An error downloading and an error extracting are two different things, but both indicate an issue downloading the file.  Are you on a reliable internet/wifi connection?  Because it sounds like something is interrupting the download.

SillyDaddy1, 2018-07-04:
My BAD... The error msg is:
//Error downloading cbfstool; cannot proceed.
Unable to download cbfstool utility; cannot continue//

thyrlian, 2019-03-22:
Same here.

After checking the source code, I found out that:

`util_source` is defined as [`"https://www.mrchromebox.tech/files/util/"`](https://github.com/MrChromebox/scripts/blob/master/sources.sh#L9) with a trailing slash.

However, in the download, it's stated as:
[`curl -sLO "${util_source}"/cbfstool.tar.gz`](https://github.com/MrChromebox/scripts/blob/master/functions.sh#L172)
So it would be like this: `curl -sLO https://www.mrchromebox.tech/files/util//cbfstool.tar.gz`
There is redundant slash in the url.

Plus, there are two more places:
https://github.com/MrChromebox/scripts/blob/master/functions.sh#L225
https://github.com/MrChromebox/scripts/blob/master/functions.sh#L257

MrChromebox, 2019-03-22:
@thyrlian while it may not be ideal, the double slashes are not problematic and have nothing to do with the download failing

thyrlian, 2019-03-22:
Thanks @MrChromebox for the reply.  I think so too.  Otherwise other people would not succeed flashing the ROM.  And it worked for me once too.  (I think curl might be tolerant for that, but I failed if I have redundant slash on macos, so it will be good to fix it, be more consistent with other url parts)

Speaking of my case, it worked for me for the very first time, I saw the menu, but I was not sure what to choose so I decided to quit.  However, after that, I got always failure, either with downloading or extracting `cbfstool`.

I found another thread talking about the `/tmp/boot/` directory issue (posted 2 years ago).  I've checked mine, I couldn't remove them due to in use, and after each restart it's empty again.  Don't know if it's related or not.  According to that thread, I believe that you've fixed the issue (checking the space).

Anyway, I don't know why, but I failed running the script, always got stuck at downloading / extracting `cbfstool`.  Any help is more than appreciated.  Thanks.

MrChromebox, 2019-03-22:
@thyrlian I'll definitely fix it at some point, given how trivial it is.

what device/OS are you having issues with?  Under ChromeOS, partition 12 of the internal storage is mounted at /tmp/boot, and a util directory is created (util) under that in order to facilitate running cbfstool (since other partitions are mounted no-execute).  You could try running the script with -x (sudo bash -x firmware-util.sh) to see exactly where it's failing, and/or remove any output redirection that's suppressing the actual error output

thyrlian, 2019-03-22:
@MrChromebox  thanks for the tips.
Mine is: Chromebook Pixel 2013 (Link) - LINK WISTERIA AGZ-K 7175

I'll run with `-x` flag and dump the logs.  Thanks.

thyrlian, 2019-03-22:
Just got the logs:

```console
+ mount /dev/sda12 /tmp/boot
+ '[' 0 -ne 0 ']'
+ mkdir /tmp/boot/util
+ cd /tmp/boot/util
+ curl -sLO https://www.mrchromebox.tech/files/util//cbfstool.tar.gz
+ '[' 56 -ne 0 ']'
+ echo_red 'Error downloading cbfstool; cannot proceed.'
+ echo -e '\E[0;31mError downloading cbfstool; cannot proceed.'
```

As you said, your script mounts `/dev/sda12`.  And the error log is indeed from downloading the cbfstool.

Thank you.

MrChromebox, 2019-03-22:
ok, I pushed an update to address the double slash issue. The -s option shouldn't be suppressing any erro output from curl, only the progress meter, so unsure why you're not getting more verbose output for the download failing.  Does the download fail if run manually from /tmp/boot/util ? (might need to be done as sudo?)

thyrlian, 2019-03-22:
I don't know why I didn't get any further verbose output apart from the two echo lines from your script instead of from curl.  But I'll give your new version a try.  (in fact, I've tried last night with my forked version which includes this "fix", but still failing with extracting the `cbfstool.tar.gz`)

thyrlian, 2019-03-22:
Just tried with your new script, from the output, the double slash is fixed, thanks.  But the cause and the logs are the same like before.

MrChromebox, 2019-03-22:
since you've forked it, I'd say try removing the 's' from the curl command (so just -LO) and see if that gives any more info

thyrlian, 2019-03-22:
Hmm, seems like the file is incomplete, something wrong with the downloading process.

With the same URL: http://www.mrchomebox.tech/files/util/cbfstool.tar.gz

On my Chromebook (with your script):
```console
du -sh cbfstool.tar.gz
46K        cbfstool.tar.gz

tar xvzf cbfstool.tar.gz
gzip:  stdin: unexpected end of file
tar: Unexpected EOF in archive
tar: Error is not recoverable: exiting now
```

On my Macbook:
```console
du -sh cbfstool.tar.gz
464K        cbfstool.tar.gz
```

MrChromebox, 2019-03-22:
is there enough free space on /tmp/boot for the file to be downloaded? There ought to be since it's a 16MB partition and basically zero used

thyrlian, 2019-03-22:
Agree with you, should not be download problem, but due to no more space to write.

thyrlian, 2019-03-22:
```
du -sh /tmp/boot/
13M
```

So < 3M left, but `cbfstool.tar.gz` before uncompress is less than 0.5M, I wonder why...

JohnRSim, 2020-01-26:
What did you guys do to fix this I'm having the same issue.

thyrlian, 2020-01-26:
@JohnRSim, I still have this issue.

JohnRSim, 2020-01-26:
I was able to get it to work by re-running the command a number of times. I then ran into an issue with installing the ROM and kept running the same command until it worked. On pixel 2013.

MrChromebox, 2020-01-27:
99% of the time, I tell people to run a ChromeOS recovery and then retry, and then have no further issues

jimmux, 2020-02-16:
Something perhaps worth noting for anyone else getting this issue.

I tried running several times with no luck, even after ChromeOS recovery. If I downloaded cbfstool with 

`curl -LO https://www.mrchromebox.tech/files/util//cbfstool.tar.gz`

I got SSL errors. Came back to it the next day and it just worked. Is possible that there are intermittent SSL issues, or is this likely to be resolved now?

MrChromebox, 2020-02-16:
SSL cert was expired yesterday, fixed now

duranimarcos, 2021-03-06:
I get the following error. I checked and have space on drive:

`Downloading supporting files...
tar: cbfstool: Wrote only 8704 of 10240 bytes
tar: Exiting with failure status due to previous errors
Error extracting cbfstool; cannot proceed.

Unable to download cbfstool utility; cannot continue`

MrChromebox, 2021-03-06:
@duranimarcos a little more detail please - device, OS, etc

JanusK22, 2021-07-31:
I have the same issue i think that is because of rerooting i try to dowload the file many times and sometimes it works sometimes no . I think that with another link it will be better . you can create your own link and by the script just modifiying  this with your link

_https://www.mrchromebox.tech/files/util/cbfstool.tar.gz_

MrChromebox, 2021-07-31:
@JanusK22 that doesn't make any sense, what is "rerooting" anyway? The script cleans up after itself so the file is downloaded fresh each time. While it can fail due to lack of space, I've made changes to the script to try and mitigate that

JanusK22, 2021-07-31:
@MrChromebox  i think that sometimes it is not possible to download the file cause of page redirection , i dont know why but i tried installing the script with almost 7 MB in tmp/boot but there was always this error by pixel chromebook 2013

MrChromebox, 2021-07-31:
what page redirection? it's a direct link to all the files other than the initial script download