MrChromebox/firmware #101: Asus C202SA (TERRA) does not detect any SD cards on boot

GyKim91, 2018-08-03:
I am using the current latest (08/02/2018) Full ROM firmware on the device and it does not detect, list or boot from any sized or formatted SD card in the internal card reader. The image below is what the EFI 'map' output is in all of the cases (the internal storage is wiped and no other storage is connected):
[image]
I've tried using rEFInd but that does not see any SD cards either. I have tested sizes 16, 32, 64 and 128GB from Sandisk, 16GB from Kingston and an 8GB microSD with an adapter. All the cards booted fine on another laptop and an iMac regardless of whether they held Ubuntu or W10 as WindowsToGo. Using an external USB card reader works as intended. The output of 'drivers' below:
[image]
The output of 'devices' below:
[image]
This issue existed in previous versions of the ROM as well.
Your work is **amazing** regardless, keep up the great job!

MrChromebox, 2018-08-04:
I'm not seeing any issues here with SD card detection on Cyan or Celes, so it's not a platform-wide issue.  I'll have to take a closer look at Terra, since a cursory examination shows the SD reader implementation to be identical to other Braswell Chromebooks

MrChromebox, 2018-08-04:
Can you confirm that the SD card is detected by the OS once booted? @reynhout tested on his Terra and the SD card was detected in Boot Manager and block devices listed in the EFI shell

GyKim91, 2018-08-04:
I'm starting to think it may be a hardware issue, I had a case of unexplained data corruption in the past but thought it was the card's fault and had it replaced. Since then I haven't used SD cards in it much and had no issues but I did some testing now by installing Ubuntu and W10 using the Chromebook through the internal reader (previously I prepared them on another, much faster laptop), booting the cards using a USB card reader (which worked fine before, internal reader still doesn't see anything during boot) and found that this way 1/3 cases Ubuntu was not bootable and 2/3 W10 failed to boot saying "Inaccessable Boot Device"... In the failed cases the cards didn't boot on another laptop either. If I could get some steps that are known to give a visible SD card during boot then I could confirm whether the card reader is indeed malfunctioning and my case is just a unique one.

MrChromebox, 2018-08-04:
> If I could get some steps that are known to give a visible SD card during boot then I could confirm whether the card reader is indeed malfunctioning and my case is just a unique one.

any inserted SD card, bootable or not, should show up in both Boot Manager and the EFI shell's block device list/mapping.  If not, there's a hardware issue of some sort, be it the reader, the SD card, or a compatibility issue that's causing initialization to time out

GyKim91, 2018-08-05:
I solved it 😄 
I installed Ubuntu 16.04 on the internal disk and have the USB live disk plugged in, but instead of letting it boot I bring up GRUB and choose "System Setup" or use "Reset System" in the UEFI, the laptop restarts and the SD reader is visible and bootable. Could this mean that the internal SD reader times out during boot and is ignored? If the live disk is not plugged in it rarely appears in the EFI shell.
[image]
 Below is what Ubuntu "Disks" show after booting the SD card through the internal reader. W10 starts booting but eventually give the "Inaccessable Boot Device" blue screen, I'll figure out a fix for that later. The reader stays functional after restarting but if the laptop is turned off then the above solution has to be done every time its turned on again, otherwise the SD reader does not appear.
[image]
[image]

GyKim91, 2018-08-05:
I'd note that this is a "rev 2" version of the C202SA, perhaps there's enough difference from rev 1 to cause firmware compatibility issues?
[image]

MrChromebox, 2018-08-05:
> I'd note that this is a "rev 2" version of the C202SA, perhaps there's enough difference from rev 1 to cause firmware compatibility issues?

no.  All ChromeOS devices use board names for identification/disambiguation.  All Terra models use the same firmware, same OS, etc.  While there may be differences at the board level, those are all handled by a single firmware image. 

it seems the issue on your device is just a detection timeout, and adding another boot device gives the SD card enough time to init/respond.  I've also heard that for some devices which are problematic on cold boot, that going into Boot Maintenance Manager and then choosing 'Reset System' will result in them being detected on the next boot, but I have no 1st hand experience with that as none of the USB/SD media I have is problematic

GyKim91, 2018-08-05:
Windows 10 works now as intended off SD. I remember reading in the past about the card reader in the C202, being slow to recognise cards or disappointing speeds (even in ChromeOS), or as in my case straight up corrupting cards marked as "V30" (guaranteed minimum write speed). I only use [SanDisk Extreme Pros](https://www.sandisk.com/home/memory-cards/sd-cards/extremepro-sd-uhs-i) so card speed should not be holding anything back, I don't know about their "responsiveness" though 🤔  I guess this issue will be resolved fully by setting the boot timeout in the UEFI to a higher number and being able to save the setting through sessions. Thanks for the help, I can finally have both USB ports free and don't need to lug around a drive anymore to run W10 😄 Keep on rocking 🤘

closed 2018-08-05 by GyKim91