MrChromebox/firmware #103: Legacy boot menu does not boot from SD on Terra

lewellyn, 2018-08-21:
On my Terra, the SeaBIOS legacy firmware seems unable to detect any SD cards and thereby present them as boot options. I discovered this in the most unfortunate of ways: I tried to install Linux onto my SD card and could find no way to boot it.

As I want to do the following, replacing the full firmware is out (unless there's some new development I've missed):

1. Dual boot with ChromeOS.
2. Not void my warranty by removing the write protect screw.
3. Choose target OS by quick and easy mnemonic input at boot. (`L` for Linux; `D` for the ChromeOS "Desktop").

This differs from #76 in that I cannot get it to recognize any SD cards, all the way down to 2GB (non-HC/XC), or at least present them in the boot menu. The cards are recognized by ChromeOS and mounted as mmc1/mmcblk1.

Of course, my wish would be for the SD card to be prioritized with the USB stick in the boot order.

I'd really rather not lose one of my two USB ports for Linux, I don't want to use my small eMMC, and if I modify chrx to make a small partition on mmcblk0p7 I haven't yet figured out how to use that as a bootable partition for an OS on my SD card (as a workaround).

MrChromebox, 2018-08-22:
unfortunately, we're at the mercy of the hardware init done by the stock firmware and then by SeaBIOS.  Certain devices seem to have issues with SD Card detection; I've provided debug logs to the developers, patches have been tested, but no improvement has been made.  At this point, either booting off internal storage or a USB flash drive is your best bet.

lewellyn, 2018-08-22:
That is indeed unfortunate. I already confirmed there is nothing of value from `cbmem`. I also assume that your configuration of `etc/sdcard?` to manually specify the location of my mmc1 has been is correct, lacking any evidence otherwise. :(

So, I'm gathering that it's time to poke at things using the eMMC and put the pieces back together if (when) it breaks, until I manage to get an installed OS result I'm happy with. Oh well. Thanks. :)

Closing as there is nothing actionable.

closed 2018-08-22 by lewellyn

MrChromebox, 2018-08-24:
I saw a patch on the SeaBIOS mailing list for increasing the SD card power on time which appeared to help another Chromebook user, drop me an email at mrchromebox @ gmail and I'll build an update for you to test out

MrChromebox, 2018-09-12:
see if the new 2018.09.12 release fixes the issue