MrChromebox/firmware #104: No Internal Storage @ Boot | ASUS Chromebox 3 (CN65) | TEEMO

zalaare, 2018-08-25:
Using the Full UEFI ROM, their is no internal storage device found on boot.  The result is being dropped to EFI Shell.  Exiting the shell and into the EFI Interface, will show no drives/disks/devices to boot from.

Since the EFI interface /does/ see USB devices and SD cards, I am using a small SD card as a workaround for now.  Hopefully in the future I can expect to boot directly from the internal storage as it is MUCH faster than the SD card.

Thanks for all the work you do!

MrChromebox, 2018-08-25:
so the internal SSD isn't listed under Boot Manager at all?

zalaare, 2018-08-26:
Correct.

MrChromebox, 2018-08-26:
I'm at a loss to explain that.  I've not seen any issues with SATA SSD detection on any devices, definitely not on my test boxes here, and several others I know have no problem on their CN65's. And unfortunately there's no debug output from Tianocore (on production devices without a debug header) so not much I can do to troubleshoot.  Perhaps try removing and reseating the SSD in case it's a hardware/connection issue?

ReddestDream, 2018-08-26:
Also try a full power cycle/unplug power. Sometimes SSD get a bad state that only de-powering can fix.

zalaare, 2018-08-26:
I'll do attempt both solutions.  I'm thinking of restoring the original firmware and starting over as well...

zalaare, 2018-08-26:
Reseating the SSD did nothing for me.  Nor did power-cycling.  Lastly reverting to stock allows me to use the Legacy Seabios to boot from internal, but again flashing UEFI Full ROM, the Internal SSD no simply is nowhere to be found in the EFI Interface.

Perhaps it needs to manually mapped or something?  I'm out of ideas.

MrChromebox, 2018-08-26:
@zalaare what make/model/capacity is the SSD in your device?

there's nothing to manually map, either the SATA driver detects and initializes the drive or it doesn't.  In the latter case, there's no way to debug why due to Tianocore not logging to the coreboot console (it only logs to a serial port, which retail devices lack).  But this is the first and only case of an internal SATA drive failing to be recognized by Tianocore.  

Since Legacy Boot mode/SeaBIOS works, then maybe stick with that.  Or I can give you a newer UEFI build to try, but unlikely to make a difference

zalaare, 2018-08-26:
Internal SSD:
`[    2.699621] scsi 1:0:0:0: Direct-Access     ATA      TS32GMTS400      5CE  PQ: 0 ANSI: 5`
`[    2.699970] sd 1:0:0:0: Attached scsi generic sg0 type 0`
`[    2.700034] sd 1:0:0:0: [sda] 62533296 512-byte logical blocks: (32.0 GB/29.8 GiB)`
`[    2.700060] sd 1:0:0:0: [sda] Write Protect is off`
`[    2.700065] sd 1:0:0:0: [sda] Mode Sense: 00 3a 00 00`
`[    2.700100] sd 1:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA`
`[    2.702688]  sda: sda1 sda2`
`[    2.703201] sd 1:0:0:0: [sda] Attached SCSI disk`
I wouldn't mind trying a new UEFI build if it has the possibility of working.  If I can remember how to kill the evil ChromeOS White screen prior to Seabios boot I'd simply use Legacy.  As it is, its not the end of the world to boot from the sdcard and then use the internal drive for LibreELEC storage.  Not writing going on on the sdcard anyway except for upgrades.

MrChromebox, 2018-08-26:
same drive that's in my CN65, which is under my TV running LibreELEC right now.

IIRC, resume from suspend in legacy boot mode still sends you to recovery mode with the crossystem flag reset, so that's kind of a pain.  Drop me an email at mrchromebox at gmail and I'll link you to a newer build to try

zalaare, 2018-08-27:
Oh well.  For the record, The newer build didn't change anything for me.

MrChromebox, 2018-08-27:
do you have another m.2 SATA SSD you can try as a test?

zalaare, 2018-08-27:
Unfortunately, no.  I also thought about giving another SSD a try, but I have only a single other M.2 SSD and it is in my main Windows Gaming PC, but it is a full length stick and is NVMe.

I also tested if the Chromebox was having issues with the PSU.  I am using the PSU from my first Chromebox (M075U), but alas, using the PSU that came with the new box did not change the result either.

I'll keep an eye out for a M.2 short stick to try.

MrChromebox, 2018-08-27:
can you not test the SSD from your old Chromebox?  doesn't need to boot, just need to see if it's detected by Boot Manager

zalaare, 2018-08-27:
Why didn't I think of that?  I'll test this tonight when I get home from work.

MrChromebox, 2018-08-27:
if your old Chromebox has UEFI firmware and a LibreELEC setup, then the old SSD should boot right up in the CN65.

zalaare, 2018-08-27:
Okay, swapped the ssd m.2 sata cards.

CN65 + Old Sandisk = FAIL | same result | SSD not detected in EFI
M075U + New Transcend = Success | SSD detected and booted perfect

RBrang, 2018-11-20:
Same here. Full ROM Firmware coreboot_tiano-fizz-mrchromebox_20180827.rom on a CN65.

What works: Booting ubuntu install image from an usb stick and successfully installing it.
What fails: After reboot there is only the uefi shell. It does not see the internal ssd module.
Even after "exit"ing the uefi shell, there is no drive to be found in the boot options.

btw. why "tiano-fizz" and not "teemo" ? Could that be the problem?

RBrang, 2018-11-21:
I see. Thanks for the information.

It's an i3 CPU
ASUS CHROMEBOX3-N008U i3-7100U/4GB/64GB

MrChromebox, 2018-11-21:
@zalaare is your CN65 an i3/i7 model?  wonder if it's related to hyperthreading

zalaare, 2018-11-21:
`cat /proc/cpuinfo | awk -F : '/model name/ {print $2}'`
` Intel(R) Core(TM) i3-7100U CPU @ 2.40GHz`
 `Intel(R) Core(TM) i3-7100U CPU @ 2.40GHz`
 `Intel(R) Core(TM) i3-7100U CPU @ 2.40GHz`
 `Intel(R) Core(TM) i3-7100U CPU @ 2.40GHz`

MrChromebox, 2018-11-21:
ok, good data point, seems like I need to get my hands on an i3/i5/i7 box for testing then

RBrang, 2018-12-04:
Can it be that the "non-visibility" simply correlates to the disk size?
When I plug in a 4GB SD-Card, it is visible in EFI Console and I can boot from it.
When I plug in a 32GB SD-Card, it is not visible in EFI console at all and also does not show up when I press F2 and try to set boot options.

MrChromebox, 2018-12-04:
@RBrang this issue involves the internal SATA drive, nothing to do with removable SD media (which if problematic, would be a separate issue. But my 32GB card works no problem here)

Stavrosnco, 2018-12-10:
Just adding that I am also experiencing this same issue with the same CN65 (i3-7100U) Chromebox 3. External media is a workaround as I can still mount and access the drive from an external OS.

MrChromebox, 2018-12-10:
I've figured out the issue and will have a corrected build posted shortly

Stavrosnco, 2018-12-10:
That's fantastic news! Thank you so much for your work!

MrChromebox, 2018-12-10:
@Stavrosnco shoot me an email and you can give the test build a go, MrChromebox at gmail

Stavrosnco, 2018-12-10:
@MrChromebox Sent you an email, I look forward to testing it!

MrChromebox, 2018-12-26:
not going to close yet, but will be resolved in next release :)

MrChromebox, 2018-12-31:
fixed in 2018.12.30 release

closed 2018-12-31 by MrChromebox