MrChromebox/firmware #109: Swanky - Burst frequency non operational on current full UEFI firnware.

perkabalo, 2018-09-14:
**Swanky - Burst frequency non operational on current full UEFI firnware.**

CPU max frequency: (**2018.08.02**) current firmware: **2080 MHz**

CPU max frequency: (**2018.02.04**) firmware: **2582 MHz**

Issue confirmed in linux and windows.
Rollback to (2018.02.04) fully operation, as expected.

ReddestDream, 2018-11-05:
The current firmware is 2018.08.27. Please update @perkabalo.

Aside from that, I was able to confirm your findings on Squawks, another BYT Chromebook.

In my case, Windows is unable to use to burst and go above 2.17 GHz. Now, one of the issues is that it's hard to just load one core on these devices. The burst is only going to activate for one core. If both are active heavily, there will no burst. Nonetheless, I was able to create a situation where the burst should be able to activate and it does not. So, I consider this confirmed. Not sure on the cause yet. I'll likely have to bisect. Thanks for reporting.

ReddestDream, 2018-11-24:
We have discovered the cause of this issue. BYT is able to boot and mostly function w/o appropriate ucode loading. Boost however does not function w/o proper ucode. Ucode has now been fixed to load properly after it was broken between 2018.02.04 and 2018.08.02. Thx @perkabalo

MrChromebox, 2018-11-27:
I've updated all the Baytrail firmware with the fixed microcode, so updating/re-flashing will resolve the issue

closed 2018-11-27 by MrChromebox