MrChromebox/firmware #110: Acer 720 - hardware interrupts 30% CPU with UEFI

clwgg, 2018-09-15:
I recently installed AntiX Linux 17.1 on my Acer 720 Chromebook (Developer Mode, Stock Firmware). Most things worked fine, but I was struggling to fix a known issue where the system log gets flooded with ehci errors after wake-up from suspend (see [here](https://wiki.archlinux.org/index.php/Acer_C720_Chromebook#Lots_of_ehci_errors_in_dmesg_after_resume)). Most fixes one finds are based on disabling certain things with systemd (see [here](https://wiki.archlinux.org/index.php/Chrome_OS_devices#Fixing_suspend) and [here](https://wiki.archlinux.org/index.php/Acer_C720_Chromebook#Lots_of_ehci_errors_in_dmesg_after_resume)), but some [suggest](https://bugzilla.redhat.com/show_bug.cgi?id=1218734#c19) it's a firmware issue at its core.

I decided to try to update my firmware with the "Install/Update the RW_LEGACY Firmware" utility of the script. However, the problem persisted. After that, I used the "Install/Update Custom coreboot Firmware" utility to flash the custom firmware. This fixed the ehci error flood after suspend, but now, at all times, 30% of my CPU are eaten up by hardware interrupts. I saw an ArchLinux user describe a similar problem on the Arch forums [here](https://bbs.archlinux.org/viewtopic.php?pid=1798921#p1798921), also running the UEFI bootloader (Not OP of the linked post, but the reply I link to). My symptoms are the same he describes (top showing 30% CPU usage by 'hi'). I wonder if this issue is related in any way to [this one](https://github.com/MrChromebox/firmware/issues/68)?

Any help would be greatly appreciated. Please let me know if I can supply any additional information.

clwgg, 2018-09-15:
Thanks a lot for the reply. I post a couple of logs below.

$ cat /proc/interrupts (after 5min uptime)
[interrupts.log](https://github.com/MrChromebox/firmware/files/2386085/interrupts.log)

$ uname -a
Linux antix-cb 4.9.87-antix.1-amd64-smp #2 SMP PREEMPT Tue Mar 13 16:16:18 EET 2018 x86_64 GNU/Linux

$ lspci -k
[lspci.log](https://github.com/MrChromebox/firmware/files/2386086/lspci.log)

maikrhg, 2019-01-18:
I have exactly the same problem with my Acer C720.

lpersin, 2019-03-20:
Same problem here on HP Falco.

MrChromebox, 2019-03-20:
not much I'm able to do about this. It only affects certain kernels (and not the mainline kernel, best I can tell). Something with coreboot's config may be a factor, but I simply don't have the time to try and reproduce and isolate the root cause.

mauritsvdvijgh, 2019-04-30:
I am experiencing this same problem using arch linux on a peppy/C720P chromebook, on linux kernel 5.0.9 I see 25-30% hardware interrupts in top. On the lts kernel I see hardware interrupts at 0%.

cat /proc/interrupt output with some time in between on 5.0.9:

[proc-interrupts.txt](https://github.com/MrChromebox/firmware/files/3131766/proc-interrupts.txt)
[proc-interrupts-560s.txt](https://github.com/MrChromebox/firmware/files/3131765/proc-interrupts-560s.txt)

cat /proc/interrupt output with some time in between on 4.9.36:
[proc-interrupts.txt](https://github.com/MrChromebox/firmware/files/3131770/proc-interrupts.txt)
[proc-interrupts-400s.txt](https://github.com/MrChromebox/firmware/files/3131769/proc-interrupts-400s.txt)

Will provide you with more debug data/help testing or whatever that helps / if you have time for looking into this.

amfern, 2020-02-22:
I am experiencing this same problem using arch linux on a panther chromebox-M004U, on linux kernel 5.5.4 30% hardware interrupts(all on single core) in htop. On the lts/4.9.214 kernel the hardware interrupts are at 0%.

MrChromebox, 2020-02-23:
@amfern can you provide output of `cat /proc/interrupts` please

ccaapton, 2020-02-23:
I have a C720 peppy with Celeron(R) 2955U. Using "MrChromebox-4.11.1 01/10/2020" and linux kernel 5.5.1. I have the same interrupts problem mentioned above. 
[image]

This is my /proc/interrupts:
[interrupts.txt](https://github.com/MrChromebox/firmware/files/4240690/interrupts.txt)

I also have the issue #68 problem, having one of the cpu core on C0 active state even when idle, and there is no package cstate at all:
[image]

MrChromebox, 2020-02-23:
but this is only a problem with newer kernels?

amfern, 2020-02-24:
> @amfern can you provide output of `cat /proc/interrupts` please
[interrupts.txt](https://github.com/MrChromebox/firmware/files/4246586/interrupts.txt)
[image]

NickGDev, 2020-03-01:
I'm also experiencing this issue with the latest build of Cloudready after using the mrchromebox bios ( Linux localhost 4.19.65+ #2 SMP Fri Nov 22 23:42:37 UTC 2019 x86_64 Intel(R) Celeron(R) 2955U @ 1.40GHz GenuineIntel GNU/Linux )

*Also affects live Ubuntu 19.10 and Manjaro 19.0.1 on the same machine with mrchromebox bios

I used this script 01/03/20:
cd; curl -LO https://mrchromebox.tech/firmware-util.sh && sudo bash firmware-util.sh

See my post here for input info: 
https://neverware.zendesk.com/hc/en-us/community/posts/360039326354-Touchpad-issue-on-chromebook-HP-14-Falco-Cypress-Touchpad-CR-v78-4-3-

jleahy, 2020-03-15:
I tracked this down the the "gpio_lynxpoint" module, which is the GPIO driver for the Intel PCH on these Haswell machines. If you blacklist that module then the issue goes away. As far as I can tell this isn't actually caused by any IRQ, it just seems to use a lot of CPU time in an interrupt context for no real reason.

This is on 5.4.25 (LTS), as well as the current mainline. I will try to go back through the LTS kernels to see at which point the bug was introduced. I would guess there is some disagreement between this firmware and that module, although its not clear yet where the fault lies.

MrChromebox, 2020-03-15:
I'll try to take a look at the GPIO code for Lynxpoint-LP, since AFAIK it's not an issue with Broadwell-U, and the platform is 98% the same.

MrChromebox, 2020-03-15:
ok, found and fixed the issue :)

jleahy, 2020-03-15:
Aha, brilliant. I'm happy to test on my hardware if you have a new image (I have a programmer, so I'm not worried about bricking it).

MrChromebox, 2020-03-15:
already tested it here, will try to get an update out shortly

MrChromebox, 2020-03-18:
fixed in 2020.03.17 release!

closed 2020-03-18 by MrChromebox

ccaapton, 2020-03-18:
Awesome! I just flash the new firmware and powertop shows the cpu spends about 80% at C3(pc3)