MrChromebox/firmware #115: Fan on LINK

penn5, 2018-10-10:
Hi,

After the latest update, entitled 'Fix random fan stops on LINK', my fan runs significantly slower, allowing my temps to idle at 61C on the CPU. Is this intentional?

Thanks!

MrChromebox, 2018-10-10:
idling at 61*C is far too high, what firmware version, OS, kernel, etc?  I'll fire mine up here for comparison

penn5, 2018-10-10:
Well, I had idling around 45. I haven't changed anything in my setup at all; I only updated the BIOS. I'm on Ubuntu, can't remember my kernel version but it's the default one for 18.04. I didn't update anything else, only the BIOS. I'm not sure about external factors; I didn't drop it or anything. I use it as a daily device so I'm not really in a position to go back to Chrome OS. I have noticed that the fans are running much quieter, usually, at 60°c I can hear them loudly in a noisy room, but now they are barely noticeable. All in all, I think it's a fan issue, be it hardware or software. I suspect software though.

MrChromebox, 2018-10-10:
there were no changes in the fan speed or temp thresholds from the previous release, only a fix for ensuring that the fan turned on at startup and off during suspend.  And obviously the OS can override the firmware defaults.  I'll try booting a Ubuntu 18.04 ISO here, but suspect this isn't related to the firmware (EC or system)

penn5, 2018-10-10:
I had previously tried to change the fan speed from Ubuntu, it failed as the system couldn't find a fan device. This leads me to think it may not be this, but I'll check in the morning, when I'm reunited with my link (left at school)

penn5, 2018-10-11:
Well, I'm back on my link, and i've done some tests. After resuming from sleep mode, my system is at only 40C. However, during a stress test on all 4 logical cores, the temperatures peak at 67C. However, more importantly, after the stress test, the temperature doesn't drop below 57C until I put the laptop on sleep for 5 mins and let it cool itself down.

penn5, 2018-10-11:
Just an idea, but have you changed anything about the sensors in the latest release? perhaps it's all 10*C too high?
On the subject of the EC and the system changing the fans, I can't check if its the EC because in dmesg I see the error cros_ec_lpcs GOOG0004:00: EC ID not detected. That error is coming from https://github.com/torvalds/linux/blob/master/drivers/platform/chrome/cros_ec_lpc.c#L261. I suspect that if (and only if) the EC is the issue, this may be related (changed EC firmware)

MrChromebox, 2018-10-11:
nothing has changed with the sensors, fan speeds, or temp thresholds, outside of the minimum fan speed changed from 0 to 1200 rpm.  

The issue with the cros_ec_lpc driver isn't related to the EC firmware update as best I can tell, since running the UEFI firmware with the stock EC firmware produces the same issue.

> During a stress test on all 4 logical cores, the temperatures peak at 67C, similar to before the update. However, more importantly, after the stress test, the temperature doesn't drop below 57C until I put the laptop on sleep for 5 mins and let it cool itself down.

'stress -c 4' produces a consistent temp of 67-70*C, with a fan speed of ~6100 RPM -- exactly what one would expect based on the fan speed/temp thresholds.  Killing stress, the temp immediately drops to ~58*C, and then back down to 52*C within a minute or two, with the fan speed dropping to ~3000 RPM.  

I flashed my device back to the previous 20180213 release, and see the exact same results with that, so I'm highly skeptical that the issues you are seeing are in any way related to either the EC firmware or main system firmware.

penn5, 2018-10-11:
How can I measure fan speed? It doesnt show in lm-sensors

MrChromebox, 2018-10-11:
> How can I measure fan speed? It doesnt show in lm-sensors

using ectool, try https://mrchromebox.tech/files/util/ectool2.tar.gz, extract, then run with:
sudo ./ectool pwmfangetrpm

penn5, 2018-10-11:
> nothing has changed with the sensors, fan speeds, or temp thresholds, outside of the minimum fan speed changed from 0 to 1200 rpm.
>
> The issue with the cros_ec_lpc driver isn't related to the EC [...]

OK, maybe my system just needed to settle in. I'm now getting the exact same temps as you! Sorry for all the fuss; I'm convinced that there _was_ an issue, but it has definitely gone now. I can get back to my Android Studio at last (n.b. just a few hours ago, i was getting 70*c with 50% cpu usage)

closed 2018-10-11 by penn5