MrChromebox/firmware #117: Touchpad waking up Candy device

erdnaxe, 2018-10-18:
At each boot touchpad and xhci can wake my device up:
```bash
[Candy erdnaxe]# cat /proc/acpi/wakeup 
Device	S-state	  Status   Sysfs node
LID0	  S4	*enabled   platform:PNP0C0D:00
CREC	  S5	*disabled  platform:GOOG0004:00
XHCI	  S3	*enabled   pci:0000:00:14.0
TPAD	  S3	*enabled   platform:PNP0C0E:00
TSCR	  S3	*enabled   platform:PNP0C0E:01
```

I have been disabling those at each boot since I installed your firmware because:

- When I close the lid the touchpad touch the screen and can sometimes "click" while being in my bag. This is annoying because the whole device wakes up ;
- From experience keeping XHCI enabled randomly wakes the device up (even without anything plugged in).

I'm interested in learning how linux works and if you can point me to the doc then I might be able to pull request a fix.

MrChromebox, 2018-10-18:
I'm not sure I see a problem here.  If those devices are causing issues with wake, then disabling them as wake sources via the OS is the correct solution (eg, echo disabled > /proc/acpi/wakeup/XHCI).  Disabling them in firmware means no one can use them as wake sources, and would be fixing the issue at the wrong level.

erdnaxe, 2018-10-18:
As I understand, on the firmware side you only decide which devices the OS can use as a wake up sources. So I should rather look on the linux side to patch and set by default touchpad and xhci disabled.

closed 2018-10-18 by erdnaxe

MrChromebox, 2018-10-18:
correct, though I'd use the term 'configure' rather than 'patch' since there's no need to modify source code