MrChromebox/firmware #120: Memory address error with UEFI firmware

wassname, 2018-11-26:
Hi, thanks for all the work on these firmwares. I'm reporting an issue that I and one other discussed [on reddit](https://www.reddit.com/r/GalliumOS/comments/97x8ed/galliumos_keeps_crashing/e52je7n/).

When using the UEFI firmware we got OS freezes. We got error in memtest, but when we reverted to the boot_stub firmware both of these issues disappeared.

Before reverting I tried various disc checks, updates, and looking at log files (nothing).

My hardware: 
- baytrail/glimmer thinkpad yoga 11e, gen 1
- gallium os 2.1

Memtest report:
- TODO

PineappleApplePizza's hardware:
- TODO

MrChromebox, 2018-11-26:
there's an issue with the microcode for Baytrail devices in the 0827 firmware, which I suspect may be the root cause of the issue.  Shoot me an email mrchromebox at gmail and I'll provide a corrected build for you to test

wassname, 2018-11-28:
I gave it a try, but the memtest errors remain. However this may just be my laptop, as still crashes with the BOOT_STUB firmware - even though memtest passes. However I'm comparing two differen't memtest programs (one for uefi and one for boot_stub). 

In any case, I'll close this issue unless any one else can replicate it. Thanks for the help.

closed 2018-11-28 by wassname

Incog5, 2020-01-28:
[image]
Getting similar errors on Wolf, MrChromeBox 4.11.1, MemTest86 8.3 EFI.

MrChromebox, 2020-01-28:
as far as I can tell, there's no actual impact to this outside of memtest showing false failures

Incog5, 2020-01-29:
Yes I agree.