MrChromebox/firmware #121: No video output on Lumpy until Linux loaded

btharper, 2018-12-01:
Converted stock fw to dev mode and installed current full rom for Lumpy (Samsung 5 550c - Sandy Bridge 2012) on 01 Dec 2018.

On boot no screen output or backlight control are visible. If left alone for a bit of time (~45s - 2 min), some Linux live USB distros bring up the screen (Lubuntu and Qubes installer both work), some don't bring up input (Gallium 2.0, 2.1, 3.1a1, Void-musl-lxqt don't output to screen). Unsure if 'failing' boot discs provide any output, only feedback is that power button transitions from immediate power off to requiring hold down for several seconds on Void Linux boot medium.

* Lubuntu begins screen output directly to the window manager (X11 or Wayland I assume).
* Qubes beings with init system logs (possibly with framebuffer change, graphics transition from default to driver control start) but also finishes booting to graphical environment.
* Void (void-live-x38_64-musl-lxqt) transitions to power button not immediately powering off
* Gallium (2.0, 2.1, 3.0a1) all produce no noticeable change, possibly error or grub screen.

btharper, 2018-12-02:
Tested an older version from @MrChromebox that proved successful. GalliumOS was indeed pausing at the Grub shell similarly to GalliumOS/galliumos-distro#342. Got everything installed with the alternate version.

MrChromebox, 2018-12-31:
should be fixed in 2018.12.30 release, but will wait for confirmation before closing

btharper, 2018-12-31:
Sorry for the delay in testing, the bug is still present in the 2018-12-30 release

MrChromebox, 2019-01-01:
fixed in 2018.12.31 update :)  thanks for assistance in debugging @btharper

closed 2019-01-01 by MrChromebox