MrChromebox/firmware #123: Bricked Asus Chromebox 3 N020U

remoteboat, 2018-12-18:
I ran the Install/Update RW_LEGACY Firmware option and found I could boot to chome dev mode but not to a usb.

I doubled down.  I removed the write protect screw and ran the Full Rom Firmware option, happily entering "Y" at all the warning messages.  I restarted, and now I have a paperweight. 

Power light comes on, but nothing else happens. No signal to display, and the power button doesn't turn the device back off.

Maybe this firmware doesn't work on this device.  

Tomorrow I'll look for the flash chip on the board and order a flash programmer if I find it.

MrChromebox, 2018-12-18:
works fine on the N020U I have here, as well as the 5 others I've flashed, except for the internal SATA drive not being detected (and I have a fix for that).  The 2018 Chromeboxes use an A/B scheme for the EC RW firmware, and sometimes it gets confused after the update.  Pulling power for a few minutes sometimes helps, other times you have to pull the CMOS battery too.  

If it does need to be flashed with an external programmer, the chip is front/center on the board, easily accessible and visible.  Instructions for unbricking are on my site, along with required hardware

closed 2018-12-18 by MrChromebox

remoteboat, 2018-12-18:
I've disconnected the power, pulled the cmos battery and put it back. I
should expect to see something on the display, right?  I'm trying to
confirm it's not and issue with the usb ports not being initialized, which
is something I think I read about some where else.

MrChromebox, 2018-12-18:
you'll see the coreboot splash screen, and if no bootable OS found on internal or external storage, then it will dump you into the EFI shell.  There's no issues with the USB ports on the Kabylake Chromeboxes of which I'm aware