MrChromebox/firmware #125: i2c issues on CELES (Samsung Chromebook 3 XE500C13)

jtecca, 2018-12-27:
I believe there's an intermittent issue with the firmware that causes i2c to fail occasionally, which has a symptom of causing the trackpad driver to fail to load on a Samsung Chromebook 3 CELES.  This ticket is related to https://github.com/GalliumOS/galliumos-distro/issues/415, which is the original discussion for the trackpad symptom.  Here is the dmesg split into two blocks: the first is the error messages after the trackpad has failed and I have tried to unload/reload the trackpad driver and the second is another message related to a failure to read my SD card which I have mounted as my /home directory on this laptop:

```[10783.508434] i2c_designware 808622C1:05: timeout in disabling adapter
[10783.531863] i2c_designware 808622C1:05: timeout waiting for bus ready
[10783.531885] atmel_mxt_ts i2c-ATML0000:00: __mxt_read_reg: i2c transfer failed (-110)
[10783.561908] i2c_designware 808622C1:05: timeout waiting for bus ready
[10783.561922] atmel_mxt_ts i2c-ATML0000:00: mxt_bootloader_read: i2c recv failed (-110)
[10783.561936] atmel_mxt_ts i2c-ATML0000:00: Trying alternate bootloader address
[10783.585248] i2c_designware 808622C1:05: timeout waiting for bus ready
[10783.585266] atmel_mxt_ts i2c-ATML0000:00: mxt_bootloader_read: i2c recv failed (-110)
[10783.585786] atmel_mxt_ts: probe of i2c-ATML0000:00 failed with error -110
[10785.383302] i2c_designware 808622C1:05: timeout in disabling adapter

[11884.834029] mmc1: Tuning timeout, falling back to fixed sampling clock
[11894.853697] mmc1: Timeout waiting for hardware interrupt.
[11894.853707] mmc1: sdhci: ============ SDHCI REGISTER DUMP ===========
[11894.853715] mmc1: sdhci: Sys addr:  0x000000b0 | Version:  0x00001002
[11894.853720] mmc1: sdhci: Blk size:  0x00007200 | Blk cnt:  0x000000b0
[11894.853724] mmc1: sdhci: Argument:  0x01c861e0 | Trn mode: 0x0000002b
[11894.853729] mmc1: sdhci: Present:   0x01ff0001 | Host ctl: 0x00000017
[11894.853733] mmc1: sdhci: Power:     0x0000000f | Blk gap:  0x00000080
[11894.853737] mmc1: sdhci: Wake-up:   0x00000000 | Clock:    0x00000007
[11894.853741] mmc1: sdhci: Timeout:   0x0000000e | Int stat: 0x00000000
[11894.853745] mmc1: sdhci: Int enab:  0x02ff008b | Sig enab: 0x02ff008b
[11894.853749] mmc1: sdhci: AC12 err:  0x00000000 | Slot int: 0x00000000
[11894.853754] mmc1: sdhci: Caps:      0x0568c8b2 | Caps_1:   0x00000807
[11894.853758] mmc1: sdhci: Cmd:       0x0000193a | Max curr: 0x00000000
[11894.853762] mmc1: sdhci: Resp[0]:   0x00000900 | Resp[1]:  0x4780b55c
[11894.853767] mmc1: sdhci: Resp[2]:   0x53423634 | Resp[3]:  0x00000000
[11894.853770] mmc1: sdhci: Host ctl2: 0x0000000b
[11894.853774] mmc1: sdhci: ADMA Err:  0x00000000 | ADMA Ptr: 0x36801200
[11894.853776] mmc1: sdhci: ============================================
[11894.854125] mmcblk1: error -110 sending stop command, original cmd response 0x0, card status 0x400900
[11894.854164] mmcblk1: error -110 transferring data, sector 29909472, nr 176, cmd response 0x0, card status 0x0```

You can see that both fail with the same error code (-110).   The common thread throughout the previously linked github ticket is that this behavior only happens when running the Full Firmware for CELES  (UEFI), which makes me think that the driver for the trackpad or sdhci are fine but there's some weird driver/firmware issue here.

I'm not really a kernel hacker/debugger, so let me know what information is needed to diagnose this, otherwise I'll stumble around in the dark trying to hack the trackpad drivers to give me more information.

MrChromebox, 2018-12-27:
recent Ubuntu kernels have some config selected that causes the SDHCI errors on some ChromeOS devices; I don't see it when running LibreELEC (eg) with a 4.19/4.20 kernel whereas I do with the Ubuntu equivalent. I've also not seen any touchpad issues with my test CELES here running UEFI firmware, so not really sure how to proceed with debugging

jtecca, 2018-12-27:
@MrChromebox good to know about the Ubuntu-specific kernels.  FWIW, I'm running:

```
$ uname -a
Linux zephyr-cb 4.14.29-galliumos #1 SMP PREEMPT Sun Mar 25 03:40:24 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```

The touchpad issues only manifest with what seems to be high amounts of load, whether that be CPU or RAM (possibly going into the swap?).    I've been collecting a number of dmesg events when things go south, though I don't know how much they're all related.  This particular one came up when I was recompiling the 4.19 linux kernel repo for Gallium OS:

```
  [  419.986838] irq 9: nobody cared (try booting with the "irqpoll" option)
  [  419.986850] CPU: 0 PID: 2368 Comm: emacs Not tainted 4.14.29-galliumos #1
  [  419.986851] Hardware name: GOOGLE Celes/Celes, BIOS MrChromebox 02/04/2018
  [  419.986854] Call Trace:
  [  419.986859]  <IRQ>
  [  419.986871]  dump_stack+0x5c/0x7e
  [  419.986878]  __report_bad_irq+0x2e/0xc0
  [  419.986881]  note_interrupt+0x243/0x290
  [  419.986885]  handle_irq_event_percpu+0x41/0x50
  [  419.986888]  handle_irq_event+0x37/0x60
  [  419.986890]  handle_fasteoi_irq+0xa7/0x170
  [  419.986895]  handle_irq+0x19/0x30
  [  419.986899]  do_IRQ+0x4d/0xe0
  [  419.986902]  common_interrupt+0x7d/0x7d
  [  419.986904]  </IRQ>
  [  419.986906] RIP: 0033:0x54f0a1
  [  419.986909] RSP: 002b:00007ffdbf172ac0 EFLAGS: 00000203 ORIG_RAX: ffffffffffffffc6
  [  419.986911] RAX: 0000000004532780 RBX: 00000000001c81ae RCX: 0000000000000038
  [  419.986912] RDX: 3f00000000000000 RSI: 0000000004532400 RDI: 0000000004532400
  [  419.986914] RBP: 0000000000877e04 R08: 00000000045327e0 R09: 0000000000000000
  [  419.986915] R10: 00000000045323c0 R11: 0000000000000000 R12: 00000000045323e8
  [  419.986916] R13: 0000000000018227 R14: 0080000000000000 R15: 0000000000000001
  [  419.986919] handlers:
  [  419.986924] [<ffffffffa2436570>] acpi_irq
  [  419.986927] Disabling IRQ #9
```

Furthermore, when the trackpad has 'died', I can reliably get the 

```
10143.692965] atmel_mxt_ts i2c-ATML0000:00: Failed to read T44 and T5 (-110)
[10143.715268] i2c_designware 808622C1:05: timeout waiting for bus ready
[10143.715279] atmel_mxt_ts i2c-ATML0000:00: __mxt_read_reg: i2c transfer failed (-110)
[10143.715286] atmel_mxt_ts i2c-ATML0000:00: Failed to read T44 and T5 (-110)
[10143.744973] i2c_designware 808622C1:05: timeout waiting for bus ready
[10143.744984] atmel_mxt_ts i2c-ATML0000:00: __mxt_write_reg: i2c send failed (-110)
[10143.864252] i2c_designware 808622C1:05: timeout waiting for bus ready
[10143.864262] atmel_mxt_ts i2c-ATML0000:00: __mxt_read_reg: i2c transfer failed (-110)
[10142.067417] i2c_designware 808622C1:05: timeout waiting for bus ready
[10142.067427] atmel_mxt_ts i2c-ATML0000:00: mxt_bootloader_read: i2c recv failed (-110)
[10142.067434] atmel_mxt_ts i2c-ATML0000:00: Trying alternate bootloader address
[10142.087212] i2c_designware 808622C1:05: timeout waiting for bus ready
[10142.087223] atmel_mxt_ts i2c-ATML0000:00: mxt_bootloader_read: i2c recv failed (-110)
[10142.087343] atmel_mxt_ts: probe of i2c-ATML0000:00 failed with error -110
[10145.443824] i2c_designware 808622C1:05: timeout in disabling adapter
```

messages in dmesg when trying to unload/reload the Atmel touchpad driver.  I'm not sure if i2c is more of a kernel-specific issue here or if this would be the firmware.  I can try to play around with some of the i2c options in the Gallium 4.19 kernel to see if I get anything interesting for a traceback.

MrChromebox, 2018-12-27:
I'd install a newer Ubuntu kernel using UKUU and see if the problems persist there, will at least give another data point

jtecca, 2018-12-27:
@MrChromebox okay, I'll give that a shot.

MrChromebox, 2019-01-03:
@jtecca any progress to report with a newer kernel?

jtecca, 2019-01-04:
@MrChromebox sort of -- the newer kernels don't seem to have the same stuttering issue, but seem to instead have (what I am guessing) power issues where the laptop shuts off on its own.  I'm not getting anything useful written to logs when this event happens, so I can't say for sure what it is.  I've noticed that suspend/resume has some issues, but I wonder if there's some patches in Gallium OS that have been applied to their kernel that aren't upstream (or are just upstream bugs).

I have been testing on 4.20 and 4.19.3 using 'ukuu'.  

I've been planning to do some deeper stress testing today, so hopefully I'll get some better reports.

jtecca, 2019-01-07:
For something unrelated to the trackpad/i2c issues, I had backed up my $HOME and wiped my computer and reinstalled with the latest Gallium OS image.  I haven't been able to reproduce this issue since I've restored on the `testing` Gallium kernel.  I haven't updated the firmware since I restored, so my theory of this being a firmware issue seems to be fizzling.  

What I see when the system is at a high load (50.00+ load average in the last second according to `top`) is that ACPI kicks in and throttles the system down, which has the side effect of producing an intermittent i915 failure, but the trackpad and keyboard now never fully die -- they just stutter under the weight of the load on the system.

I kept testing on 4.20 and 4.19.3 kernels built from `ukuu`, and despite some suspend/restore issues (which I think are patched downstream by Gallium or just different defaults from the stock kernels) and still wasn't able to repro my initial issue all using the same stock firmware.

All this to say is that this points to not being a firmware issue, so I'll close this issue unless I find something later on that clearly points to it being a firmware issue.

closed 2019-01-07 by jtecca