MrChromebox/firmware #126: Aaron_Yuna weird high System CPU usage under windows

igel-itzo, 2019-01-01:
Noticed this with the latest Firmware (2018-12-30)
Windows 8.1 Pro x64 fairly recent on updates
The System process will use on average 50% of the CPU, other times it will use just 0-1% which is the norm. 
After a bit of digging with xperf I identified it coming from CmBatt.sys!CmBattSendDownStreamIrp (there are a bunch of ACPI.sys calls further in the stack). When I disable the  Microsoft ACPI-Compliant Control Method Battery the CPU utilization from system goes to 0-1%. Once I knew it was battery and ACPI related I tried Lubuntu that did not exhibit the issue (no ACPI related CPU usage), so it is probably not the hardware going faulty, though Nirsoft BatteryInfoView is unable to retrieve the battery temperature under Windows and I don't know for sure if the battery has a thermal sensor. 
The issue is observed only when the battery is charging or discharging. When the battery is fully charged and with the power adapted plugged in and supplying power, the issue isn't observed, even with the    Microsoft ACPI-Compliant Control Method Battery device/service enabled. 
Is it possible that the battery temperature is no longer properly reported by the new firmware?

ReddestDream, 2019-01-03:
Sounds related to this issue. I noticed some AC power status issues during testing, but I thought it was just a fluke. Currently investigating.

https://github.com/MrChromebox/firmware/issues/127

MrChromebox, 2019-01-03:
@igel-itzo if you can test a build with the battery fix, would be very helpful. drop me and email @ mrchromebox at gmail

MrChromebox, 2019-01-05:
fixed in 2019.01.04 release

closed 2019-01-05 by MrChromebox