MrChromebox/firmware #127: Coreboot 4.9 - Battery status

duffydack, 2019-01-02:
I have an old but great condition Toshiba (Swanky).  Battery still holds a decent amount of charge, and have it on AC pretty much most of the time.  Battery reports always being at 100% while on AC

`acpi -b`
`Battery 0: Full, 100%`

After flashing update with script 2018.12.30, my battery now reports while on AC being 99% and Unknown status.

`Battery 0: Unknown, 99%`

Unplugging and running on battery shows correctly 'Discharging'.  I let it use up about 10%, plug back in and it starts to charge again, reporting it as 'Charging' as it should.  The problem occurs when it reaches 96%, it shows as Unknown status and charges up to 99% and no further, still saying Unknown.

It may be just my old battery, but I flashed back to 20181126 rom and all was normal again, battery reporting Full at 100%.

ReddestDream, 2019-01-03:
I've been able to reproduce something that appears similar to this issue. We are currently investigating.

duffydack, 2019-01-03:
I've just tested a rom from @MrChromebox that appears to have fixed the issue.
Was this just a baytrail bug, or was it a general firmware thing?  I have an Acer "Edgar" as well, but not updated yet.

ReddestDream, 2019-01-03:
Does not seem to have affected BSW, but it was more than Baytrail (I could reproduce on HSW).

MrChromebox, 2019-01-05:
fixed in 2019.01.04 release

closed 2019-01-05 by MrChromebox