MrChromebox/firmware #128: Screen goes black after unplugging charger

spy14, 2019-01-08:
Hi!
I'm running Windows 10 Pro(OS build: 17763.195) on chell and the problem is that when I unplug the charger the screen goes black. If I plug it back sometimes the screen comes back on, but not always. From what I could see is that windows does not freeze, just the screen goes black(the backlight is still on). 

Is this issue known already? Any help would be appreciated, if more info is needed I'm happy to help and/or test changes.

MrChromebox, 2019-01-08:
first I've heard of this, and not sure it's a firmware issue.  I'll see if I can replicate it on Chell here, but I have the Pentium/1080p version so might not be apples/apples

ReddestDream, 2019-01-08:
@spy14 Check in the Intel Graphics driver settings. Sometimes there's a lower refresh rate on battery setting. You'll want to make sure it's off. It doesn't play well with all displays.

Also make sure the backlight is right. Unplug-black screen can happen because Windows has the "unplugged" brightness too low.

spy14, 2019-01-08:
@ReddestDream I'll try that, after more investigation it is only happening with windows, not in the bios or in Elementary OS, so it will be some kind of windows graphics magic.