MrChromebox/firmware #129: override Board Name

blp77, 2019-01-14:
I installed full ROM on my Zako with John Lewis script, yea, that's the first one I found on the web... 
I had some write errors but eventually everything went fine, I mean I was able to install GalliumOs. 

but could not sleep due the errors and started investigating ;) 

turns out `dmidecode -s system-product-name` returns "Panther" instead "Zako" now... 
so, I suspect I reimaged my Zako with Panther ROM... and its Legacy...

MrChromebox, 
What's the best way to upgrade with Zako UEFI with your scripts? 
If I just substitiute/hardcode in functions.sh:

`device=$(dmidecode -s system-product-name | tr '[:upper:]' '[:lower:]' | sed 's/ /_/g' | awk 'NR==1{print $1}')`

with 

`device=zako`

will that be enough?  

Thank you

MrChromebox, 2019-01-14:
To do that, you would need to download/run my script, exit, edit firmware-util to not delete/re-download functions.sh, and then replace the line you suggested.  then re-run the script without re-downloading first.

Alternately, you could also: 
- run my script
- restore stock firmware, choosing Zako when prompted
- update RW_LEGACY firmware
- set boot options to legacy/1s
- reboot into GaOS
- re-run script and flash UEFI firmware

blp77, 2019-01-14:
it worked! I went for option 1. 
Additionally had to copy my RO_VPD backup file to /tmp/vpd.bin to re-store my original MAC address (lost it when re-imaged with John Lewis script). 

Also found in firmware.sh
`#TODO - user enter MAC manually?` :)

Thanks for help!

closed 2019-01-14 by blp77