MrChromebox/firmware #130: Can't boot from Internal MicroSD Card Reader (UEFI)

ProdigySim, 2019-01-21:
I'm running a Asus Chomebox 3 / CN65  (Teemo) with Firmware MrChromeBox-4.9 01/04/2019.

I can successfully boot from a USB3 microSD card reader to a linux installation disk & install the OS on a MicroSD card in the built-in MicroSD reader. The GRUB entry for the new installation appears in the UEFI BIOS menus, but will not boot. It flips to a black screen briefly, then back to the boot menu.

If I take the same MicroSD card and put it in a USB3 MicroSD card reader, the UEFI firmware boots the GRUB UEFI bios properly. So, it seems like the MicroSD card reader is not found by the UEFI firmware.

Additional info: if I try a "Boot to file" option, it appears as though none of my MicroSD card partitions appear when I use the internal MicroSD reader. When I use the USB3 MicroSD reader, they do appear.

MrChromebox, 2019-01-21:
hmm, I'll have to take a look at this, can't say I ever tested it on mine

pcs3rd, 2019-12-19:
Same issue. I have grub on internal mmc and everything else on an external drive. The only difference is that I get a grub error.

closed 2024-04-28 by MrChromebox