MrChromebox/firmware #132: TPM not quite functioning in Windows

Izzard-UK, 2019-02-19:
Hi. I see there is an existing thread "Enable/Support TPM": 
https://github.com/MrChromebox/firmware/issues/21

If I'm reading that correctly it suggests that TPM is now available on my Dell Chromebook 11 (WOLF). However, I've been unable to get it working for BitLocker on Windows.  After a lot of faffing around I've got it to the point where Windows believes "The TPM is ready for use" but initialising BootLocker fails with "An internal error has occurred within the Trusted Platform Module supporting program."  I realise you don't run Windows on any of your Chromebook but I'd love to get this working so if you're interested in looking into it I'd be happy to run tests etc.

So far I've been able to clear the TPM using tools in Ubuntu, which left the TPM cleared and deactivated.  I was subsequently able to set the owner's password and retrieve the public EK (with tpm_getpubek).  However, I don't seem to be able to fully initialise the TPM in Ubuntu either and it seems to be an issue of not being able to take ownership.  Most of the errors seem to revolve around "The target command has been disabled", implying some features are not implemented and my best guess at the moment is this is because the necessary functions aren't implemented in the full firmware.

I'd like to think this just needs something simple in the firmware but I'm probably clutching at straws there! 

**Contents of log file created by TPM troubleshooting in Windows**

********************************************************************************
*    The information within the TPM_Logs folder was generated from clicking    *
*    "Collect Logs" in Windows Defender Security Center. This is diagnostic    *
*   information that may be used by an IT administrator to troubleshoot the    *
* installed Trusted Platform Module (TPM). Please zip the folder and attach it *
*          to issues filed through Feedback Hub or with an IT admin.           *
********************************************************************************

System Information
  -Product Name: Windows 10 Pro
  -Build String: 17763.1.amd64fre.rs5_release.180914-1434
  -Base Board Manufacturer: GOOGLE
  -Base Board Product: Wolf
  -BIOS Release Date: 01/04/2019
  -BIOS Vendor: coreboot
  -BIOS Version: MrChromebox-4.9
  -System Family: Google_Slippy
  -System Manufacturer: GOOGLE
  -System Product Name: Wolf
  -System SKU: 

TPM Device Information
  -TPM Present: True
  -TPM Version: 1.2
  -TPM Manufacturer ID: IFX
  -TPM Manufacturer Version: 4.32
  -PPI Spec Version: 1.2
  -Ready For Storage: True
  -Ready For Attestation: False
  -Information Flags: 0x00000007
  -Is Clear Possible: True
  -Is Capable For Attestation: False
  -Clear Needed To Recover: True
  -TPM Has Vulnerable FW: True
  -TPM FW Vulnerability: 0x00000001
      -ADV170012 - IFX ROCA/Riemann
  -PCR7 Binding State: 0
  -TPM Spec Level: 2
  -TPM Spec Revision: 3

**Error message logged whenever Windows tries to initialise the TPM**

The Trusted Platform Module (TPM) hardware on this computer cannot be provisioned for use automatically.  To set up the TPM interactively use the TPM management console (Start->tpm.msc) and use the action to make the TPM ready.

Error: The TPM provisioning action is not allowed by the specified flags.  For provisioning to be successful, one of several actions may be required.  The TPM management console (tpm.msc) action to make the TPM Ready may help.  For further information, see the documentation for the Win32_Tpm WMI method 'Provision'.  (The actions that may be required include importing the TPM Owner Authorization value into the system, calling the Win32_Tpm WMI method for provisioning the TPM and specifying TRUE for either 'ForceClear_Allowed' or 'PhysicalPresencePrompts_Allowed' (as indicated by the value returned in the Additional Information), or enabling the TPM in the system BIOS.)
Additional Information: 0x80010c

MrChromebox, 2019-02-20:
I'm working on getting the TPM fully functional with Tianocore, enabling secure boot, bitlocker, etc.  There's a few things missing on the coreboot side which Windows needs but Linux doesn't

Izzard-UK, 2019-02-20:
Ah, that’s great news! In which case I will stop banging my head against the wall with it. Thank you for your work. I’ll be very happy to try to assist as and when you have something that can be tested.

Hariiiiiiii, 2019-03-30:
I know a previous issue with the TPM involved failure to initialize properly from within coreboot, I'm not sure if that's changed since, but I found a publication where a group encountered what sounds like the exact same issue and was able to fix it:

https://ecc2017.coreboot.org/uploads/talk/presentation/37/Enabling_TPM2.pdf

MrChromebox, 2019-03-30:
@Hariiiiiiii as per the title of the presentation, that's about TPM 2.0 (not TPM 1.2, which is what the vast majority of devices I support use).  It's also not an issue of coreboot setting things up, but the payload exposing the bits that Windows is looking for on a UEFI system.  It's Tianocore that needs TPM support added.

Hariiiiiiii, 2019-03-30:
@MrChromebox 

Ah, gotcha, too bad then. I'm sure you'll figure it out soon enough. Thanks for your continued support of all of these devices!

Hariiiiiiii, 2019-04-05:
@MrChromebox 

You may be way ahead of me on this, but from what I can see, the TPM is ending up in a blocked/locked out state but then cannot be cleared, reset, or managed by the OS. The problem ultimately comes down to an issue with BIOS configuration relating to a physical presence command. This is the error I'm getting:

https://support.microsoft.com/en-us/help/2561178/clear-tpm-fails-with-error-code-0x80290300

Izzard-UK, 2019-04-05:
That's what I concluded too. In the end I used Bitlocker without the TPM by removing the Windows TPM driver (otherwise it insisted on trying - and failing - to use the "broken" TPM).

Hariiiiiiii, 2019-04-05:
I did some digging in the code and I found this:

https://github.com/tianocore/edk2/blob/7367cc6c24d01b400d2370ffd58ae02854a56b32/SecurityPkg/Tcg/PhysicalPresencePei/PhysicalPresencePei.c

the flag "PcdTpmPhysicalPresence" is always set to true in the link below, but it doesn't matter because it only checks against a false value:

https://github.com/tianocore/edk2/blob/8a9301cdd75439e781754014f514fa06d99140d1/SecurityPkg/SecurityPkg.dec

So then it checks if there is a pending TPM request, and if not, then it locks physical presence by default. I bet if physical presence is set to be enabled by default as per the first link, then the OS should be able to go through with clearing the TPM and creating a new ownerauthorization token etc etc etc

MrChromebox, 2019-04-05:
@Hariiiiiiii I don't believe either of this packages is currently included in the CorebootPayloadPackage Tianocore build (ie, what gets built for use with coreboot); AFAIK the TPM isn't being touched by Tianocore currently

Hariiiiiiii, 2019-04-05:
@MrChromebox ah you're right, I didn't realize you were only building with the subset of Tianocore's libraries listed here: 

https://github.com/MrChromebox/edk2/blob/126d2aaa91be1cfaea969f562df09f59421b10d3/CorebootPayloadPkg/CorebootPayloadPkgIa32X64.dsc

Well I guess then the solution would almost definitely involve including some of these libraries related to TPM interface in your payload:

https://github.com/MrChromebox/edk2/blob/master/SecurityPkg/SecurityPkg.dsc

shuxin, 2019-06-01:
> @Hariiiiiiii as per the title of the presentation, that's about TPM 2.0 (not TPM 1.2, which is what the vast majority of devices I support use).  It's also not an issue of coreboot setting things [...]

[TPMFWupdate.zip](https://github.com/MrChromebox/firmware/files/3243420/TPMFWupdate.zip)
This is the Infineon TPM chip Flashing tool. 
there is a firmware called (TPM12_4.43.257.0_to_TPM20_5.62.3126.0.BIN)

MrChromebox, 2019-06-01:
@shuxin again, not an issue of 1.2 vs 2.0; the issue is that Tianocore doesn't expose the TPM to Windows in the way it needs to be usable.  It's on the to-do list.

ryanm816, 2021-06-29:
@MrChromebox You may want to revisit this issue soon as Windows 11 requires both TPM 2.0 and Secure Boot to run now.  I managed to get the developer version to run on my Chromebook but it refuses to update because of secure boot issues.  This will become a much larger problem soon when MS starts distributing Windows 11 soon

https://docs.microsoft.com/en-us/windows/whats-new/windows-11-requirements

MrChromebox, 2021-06-29:
meh. Let's assume that TPM 2.0 requirement holds -- 99% of the Chromebooks which run Windows with decent functionality are TPM 1.2. 

I don't have any interest in spending my time doing things specifically for Windows, an OS I neither use nor like. If someone else gets it working, I'll integrate it into my tree, but not going to go out of my way to make Windows 11 run on 5+ year old Chromebooks.

albertofustinoni, 2021-07-20:
Came to check on this issue for the same reason as @ryanm816 and want to add that Windows 11 only requires TPM 1.2

Also, the TPM has uses beyond Windows 11: it allows for setting up disk encryption with automated unlocking without having the password in cleartext somewhere on the device (removable media or otherwise). Bitlocker is the most common way it's done but there's a way to set up LUKS and VMWare to use the TPM that way too.

I use an Asus CN60 as a host for my VMs with Ubuntu and would love to be able to encrypt the disk, since I host a domain controller in there.

MrChromebox, 2021-07-20:
I understand, I just don't have the bandwidth required. I don't get paid to work on Chromebooks/Tianocore, and supporting 100+ devices with updates and bug fixes is challenging as it is. If someone else wants to do it, I'll accept a pull request. If another Tianocore fork implements it, I'll try pulling it in, but this is not simple stuff.