MrChromebox/firmware #133: Built-in display blank after boot on BUDDY

georgebashi, 2019-02-26:
I've got a BUDDY flashed with the new UEFI firmware. Works fine in windows, but on Linux the primary display turns blank after the kernel initializes DRM. I can get the main display to work with `nomodeset`. Booting NixOS, I was able to get the main display to work by passing `video=eDP-1:1920x1080` to the kernel, but that doesn't work with Ubuntu-based distros. I think this is because the mode is set earlier by grub and passed through (or something, I wasn't able to understand what was happening there).

Anyway, @MrChromebox asked me to boot with `drm.debug=0xe` and post the output, so [here you go](https://github.com/MrChromebox/firmware/files/2903967/dmesg-drm.txt). I did that and the complete output is attached here. I should note that the HDMI port works fine the whole time (any linux, windows, etc).

Thanks for all you do! It's so awesome to be running non-ChromeOS on this machine now! ❤️

georgebashi, 2019-02-26:
Here's an updated 
[dmesg output](https://github.com/MrChromebox/firmware/files/2904161/dmesg-drm.txt). This time I'm booting ubuntu cosmic server, and passing no kernel parameters.

MrChromebox, 2019-12-19:
Buddy has been fixed as of 2019.12.18 :)

closed 2019-12-19 by MrChromebox