MrChromebox/firmware #135: FULL ROM UEFI for ApolloLake, timeframe?

johmcc1138, 2019-03-20:
Over on the supported devices page, I see that FULL ROM UEFI is "TBD" for ApolloLake devices... any ideas on when that might come to fruition?  
Using Legacy, I can boot and install Ubuntu... but, after suspend or reboot, it wipes and comes back to ChromeOS Rescue. 

Would love to just permanently get Ubuntu installed and move forward.

Thanks for all the awesomeness here!

johmcc1138, 2019-03-20:
Also, is there anything that I could offer to help bring that to fruition?  Testing, eval, etc?

MrChromebox, 2019-03-20:
honestly, I don't see it happening at this point. Given the lack of functionality in mainline Linux (or Windows), there's not much point in a UEFI Full ROM for Apollolake devices, and it's not something I have the time to build/test currently.

johmcc1138, 2019-03-20:
Any other ROM that would allow a full install of Linux and not break on suspend or reboot? 
Or are the ApolloLake Chromebooks just hard-locked to being ChromeOS forever?  Would like to repurpose this machine, apart from ChormeOS.

MrChromebox, 2019-03-20:
nope.  There might be some way to disable the TPM though which would potentially fix the suspend issue.  I'll see what I can find on that

Dujyn, 2019-03-30:
It would be great if there was a way to fix the suspend issue. If i can help out in any way. Pleas let me know.

amg-12, 2019-08-08:
Any update on this?

MrChromebox, 2019-08-08:
still not planning to provide UEFI firmware for Apollolake. Suspend/resume issue may be fixed once driver for Google's CR50 TPM chip lands in upstream kernels

closed 2019-08-08 by MrChromebox