MrChromebox/firmware #137: Touchscreen not working - Link / Pixel 2013 - Full ROM v4.9 01/13/2019

owenrw, 2019-04-23:
@MrChromebox I'm currently experiencing an issue with my ChromeBook Pixel 2013 (Link) whereby the touchscreen doesn't appear to be working after flashing the 4.9 (01/13/2019) Full ROM.

When running evtest, the output doesn't list the touchscreen as an input device:
```
No device specified, trying to scan all of /dev/input/event*
Not running as root, no devices may be available.
Available devices:
/dev/input/event0:      Sleep Button
/dev/input/event1:      Sleep Button
/dev/input/event2:      Lid Switch
/dev/input/event3:      Power Button
/dev/input/event4:      Video Bus
/dev/input/event5:      AT Translated Set 2 keyboard
/dev/input/event6:      Atmel maXTouch Touchpad
/dev/input/event7:      HDA Intel PCH Front Line Out
/dev/input/event8:      HDA Intel PCH Line Out
/dev/input/event9:      HDA Intel PCH Front Headphone
/dev/input/event10:     HDA Intel PCH HDMI/DP,pcm=3
/dev/input/event11:     HDA Intel PCH HDMI/DP,pcm=7
/dev/input/event12:     HDA Intel PCH HDMI/DP,pcm=8
```
Looking in dmesg, the output related to the touchscreen is:
```
[    0.432236] atmel_mxt_ts 1-004b: GPIO lookup for consumer reset
[    0.432242] atmel_mxt_ts 1-004b: using lookup tables for GPIO lookup
[    0.432246] atmel_mxt_ts 1-004b: lookup for GPIO reset failed
[    0.472108] atmel_mxt_ts 1-004b: Family: 130 Variant: 1 Firmware V1.0.AA Objects: 22
[    0.526204] atmel_mxt_ts 1-004b: Direct firmware load for maxtouch.cfg failed with error -2
[    0.539079] cros_ec_lpcs GOOG0004:00: bad packet checksum f2
[    0.541110] cros_ec_lpcs GOOG0004:00: MKBP not supported
[    0.543105] cros-ec-dev cros-ec-dev.1.auto: cannot get EC features: 0/1
[    0.543125] cros-ec-dev cros-ec-dev.1.auto: EC does not have accelerometers.
[    0.550099] atmel_mxt_ts 1-004b: Touchscreen size X2040Y1360
```
...and the touchscreen isn't added as an input device, instead the next line adds the trackpad as input6, instead of the expected assignment to the touchscreen.

I'm using Arnold the Bat Chromium Vanilla image, however I believe this is a problem related to the firmware, as another user has the same installation with the exception that they are using firmware 08/27/2018.

There is an issue discussing this here: [https://github.com/arnoldthebat/chromiumos/issues/261](https://github.com/arnoldthebat/chromiumos/issues/261)

Is firmware 08/27/2018 available for download still, so I can try flashing this and see if it resolves the issue, (or if you have any thoughts or suggestions? =)

MrChromebox, 2019-04-24:
this isn't a firmware issue, working fine here on my 2013 Pixel with 2019/01/13 firmware and Linux Mint 19.1, kernel 4.15.0-43.

closed 2019-04-24 by MrChromebox

owenrw, 2019-04-25:
@MrChromebox I flashed a copy of the 2018/08/27 firmware and the touchscreen now works fine in Chromium.

With 2019/01/13 the touchscreen worked fine in both Mint and Elementary but not in any version of Chromium stretching back to r67.

I appreciate that this may not be a bug in the firmware, but there is definitely a change between the latest firmware version and the 08/27 firmware which may be exposing either a bug in the Chromium driver or is not exhibiting the behavior it expects.

I know, like all of us, you're busy and you look after this in your non-work time, however I'd like to raise this with the appropriate team or person to (try) and have this resolved as I don't believe this will only be affecting me, so if you could provide any assistance, that would be appreciated (at least by me).

MrChromebox, 2019-04-25:
hard to say given the driver doesn't give any useful info when it fails. Also, the snippet you posted is for the touchpad (i2c address 4b), not touchscreen (i2c address 4a), unless the driver is reporting them backwards for some reason. I suspect it's due to the driver and change to the 4.14 kernel, but there no obvious changes on the firmware side between the 8/27 and 1/13 releases which would account for this issue.  The touchscreen code literally hasn't changed since it was added in 2013.

> I know, like all of us, you're busy and you look after this in your non-work time, however I'd like to raise this with the appropriate team or person to (try) and have this resolved as I don't [...]

I'm not sure who you think supports this besides me -- it's a one-man operation.

MrChromebox, 2019-04-25:
started to think about this some more, what else would affect the i2c bus, and remembered that LINK weirdly has the i2c bus routed thru the GPU. Then I remembered that for the 4.9 release, I changed all the SNB/IVB devices over to using coreboot's native graphics init.  So I did a test build with the old GOP driver init, and just like that the touchscreen is back.  Makes no sense that the touchpad would work but not the touchscreen, or that it would only affect ChromiumOS, but there we have it.

I'll have this fixed for the upcoming 4.10 release

owenrw, 2019-04-25:
> I'm not sure who you think supports this besides me -- it's a one-man operation.

In reference to this, I meant the Chromium touchscreen driver... I.e. I'd raise the issue with the team/person that supports this, not your firmware. I'm aware you're only one person =)

> I'll have this fixed for the upcoming 4.10 release

Thank you for taking the time to look at it! I definitely appreciate it.