MrChromebox/firmware #138: Peppy: Timeout setting tianocore is not saved

mauritsvdvijgh, 2019-04-30:
It seems that the timeout setting is not persistent on my peppy/Acer C720P running the MrChromebox-4.9 01/04/2019 firmware. Setting it using efibootmgr also does not seem to work

Is this supposed to work? 
If it is not supposed to work is there some way I can set the timeout to 0?

This seems similar to: https://github.com/MrChromebox/firmware/issues/10
But that issue is closed and some settings do save like the boot entries, order and next boot value.

MrChromebox, 2019-04-30:
saving of the timeout is not yet implemented. Only way to adjust is to change and recompile

mauritsvdvijgh, 2019-04-30:
I figured as much, will attempt to do so, thanks for the quick reply!

MrChromebox, 2019-11-26:
timeout setting has been removed from the UI, so technically this is no longer an issue :)

closed 2019-11-26 by MrChromebox

mauritsvdvijgh, 2019-11-26:
alright then :')