MrChromebox/firmware #140: Add support for SONA (HP 14" x360 G1)

jmandawg, 2019-05-06:
Any chance you can add support for SONA?  Thanks.

jmandawg, 2019-05-07:
So I installed your legacy seabios from your script and it seems to work great.  

I have one question though, I installed archlinux to my microSD card (with grub on the MBR).  When i boot into seabios (CTRL+L) there are 2 boot options:

```
1. MMC drive DF4064 59640MiB
2. USB MSC Drive Generic STORAGE DEVICE 0819
```

I'm assuming 1. is the chromeos from my internal storage, and 2. is the microSD card.  When i choose option 1. I see vmliunx loading then it just goes to a black screen, chromeos does not load, and i have to force off the system.  When i choose option 2. it boots into grub/archlinux successfully.

My questions are:

1. is chromeos supposed to be bootable from seabios?
2. is there a way to change the boot order in seabios or remove the first boot option in my list?

Thanks.

MrChromebox, 2019-05-07:
> is chromeos supposed to be bootable from seabios?

no

> is there a way to change the boot order in seabios or remove the first boot option in my list?

yes, re-run the RW_LEGACY firmware update, and choose 'U' for USB default when prompted

jmandawg, 2019-05-07:
Thanks, worked like a charm.

Should I keep this ticket open in hope there will be a full firmware eventually?  If not i can close it.  Thanks for the great work on this!

MrChromebox, 2019-05-07:
I don't want to say never, but I don't have any plans to release firmware for newer ChromeOS devices at this time

jmandawg, 2019-05-07:
Ok, i'll just close it then, but i'll keep an eye on your webpage just in case.  Thanks.

closed 2019-05-07 by jmandawg