MrChromebox/firmware #142: Does UEFI full Rom support boot from sd card for my asus c302 skylake m3?

thomasyonug, 2019-05-24:
If not, can I boot from a usb convert to type-c device?

MrChromebox, 2019-05-24:
I believe booting from an SD card should work on that device, but have not personally tested.

any USB/USB-C attached device shoould work without issue, provided it contains an EFI bootloader

thomasyonug, 2019-05-26:
> I believe booting from an SD card should work on that device, but have not personally tested.
> 
> any USB/USB-C attached device should work without issue, provided it contains an EFI bootloader

Sry guy. I tried to boot from sd card but can't detect it. So I have to buy a converter to install my os.

closed 2019-05-26 by thomasyonug

ralphmapper, 2020-11-17:
[image]
I definitely booted from an SD card and installed cloud ready.   So I think it is possible.