MrChromebox/firmware #145: the minipcie on PANTHER(asus cn60) does not support asm1061

shuxin, 2019-06-06:
I have bought a minipcie-sata adapter( asm1601) replaced the wifi chip. but it cannot be detected by os.

MrChromebox, 2019-06-06:
it's not going to work, that socket is only wired for PCIe and USB, not for SATA

closed 2019-06-06 by MrChromebox

shuxin, 2019-06-06:
but, a minipcie-usb3(nec d720202) works well.

MrChromebox, 2019-06-06:
exactly -- as I just said, the socket is wired for PCIe (for wifi) and USB (for BT). The pins for SATA functionality are not connected to the SoC, there is no way to make SATA work for it.

shuxin, 2019-06-06:
[image]
the left is asm1601, and not works. the right is nec d720202, works well.
It's a standard PCIE 2.0 x 1, I think there must be a way to make it works.

shuxin, 2019-06-06:
> exactly -- as I just said, the socket is wired for PCIe (for wifi) and USB (for BT). The pins for SATA functionality are not connected to the SoC, there is no way to make SATA work for it.

I know, it not supported msata. the asm1601 is a standard PCIE 2.0 x 1 device.

MrChromebox, 2019-06-06:
I'm afraid not, the required pins simply aren't connected on the socket

shuxin, 2019-06-06:
The pcie pins are all connected, but some drivers code lost in coreboot and edk2.
like this: PlatformDxe/Asmedia118x.c http://patches.linaro.org/patch/125638/

shuxin, 2019-06-06:
https://github.com/ms-iot/imx-edk2-platforms/blob/df26792b79c25be0fc278ca5d8824bb563c05361/Silicon/Socionext/SynQuacer/Drivers/PlatformDxe/Pci.c

shuxin, 2019-06-06:
https://github.com/ms-iot/imx-edk2-platforms/blob/df26792b79c25be0fc278ca5d8824bb563c05361/Silicon/Socionext/SynQuacer/Drivers/PlatformDxe/Pci.c
Can it be included by the next version?