MrChromebox/firmware #147: Kabylake - ALKALI360

metaquanta, 2019-06-10:
I have one. Can I help test?

I didn't see any other issues RE Kabylake TBDs

MrChromebox, 2019-06-11:
I already have support for this model, will be included in next release
(coreboot 4.10)

mspiegel31, 2019-07-27:
Thank you for providing these scripts and firmware; they seem really useful.

I just picked up a refurbished [HP Chromebook x360](https://store.hp.com/app/pdp/hp-chromebook-x360-14-da0021nr) with a Kabylake chip.  Is this model already supported as well?

I see that Coreboot 4.10 was released July 22nd; should we expect the update scripts to utilize the new coreboot release at some point in the future?  Currently I don't have the option to flash a new ROM.

[image]

MrChromebox, 2019-07-27:
> I just picked up a refurbished HP Chromebook x360 with a Kabylake chip. Is this model already supported as well?

https://mrchromebox.tech/#devices

> I see that Coreboot 4.10 was released July 22nd; should we expect the update scripts to utilize the new coreboot release at some point in the future? Currently I don't have the option to flash a new ROM.

the next update I release will be based on coreboot 4.10. I already support 60+ devices, rebasing and testing all of them takes time. UEFI Full ROM firmware for SONA and other Poppy-based Kabylake boards is planned for the 4.10 release.

mspiegel31, 2019-07-27:
🤦‍♂ I will RTFM;  somehow missed the devices page.

I will keep an eye out for that release then;  thanks for the fast response!

metaquanta, 2019-08-10:
Hello MrChromebox,
I built coreboot from your coreboot/working branch and your configs/.config.nami.eufi for AKALI360. I provided all the blobs referenced in the config. I pulled the IntelGopDriver.efi from coreboot_tiano-eve-mrchromebox_20190214.rom and the rest from my stock firmware. 

What do you think the odds are it works?

I haven't tried it yet. I'm waiting for the programmer, test-clip, etc I ordered to arrive.

MrChromebox, 2019-08-10:
1) I'm about to release updated firmware, which will include support FOR AKALI

2) you want a USB-C debug cable, not. a chip clip. see https://wiki.mrchromebox.tech/Firmware_Write_Protect 
same thing is used to flash thr firmware externally

metaquanta, 2019-08-10:
Oh. Closed-case-debugging looks way better than a chip clip! Does anyone sell the cable? I actually have a usb-c breakout hanging around somewhere, but professional cables are often worth getting.

metaquanta, 2019-08-10:
nm. found it. Thanks!

MrChromebox, 2019-08-29:
added as part of 2019.08.22 release

closed 2019-08-29 by MrChromebox