MrChromebox/firmware #148: Support UEFI on SONA please

lakano, 2019-06-13:
Hello!

Thanks for your awesome works ! It's pretty cool !

I have one week to known if I could keep a HP Chromebook x360 G1 ( core i5 8th generation, 8 GB RAM / « SONA » ), and to known that I need to be sure I could boot on UEFI devices in few days or weeks. Next week, I can return the Chromebook without any charges.

I'm working under GNU/Linux since 1998,  and since 5 years now, I've installed my Fedora on a USB key ( Corsair GTX Voyager SSD ). This permit me to easily plug my USB key on any UEFI-capable x64 devices ( at works, on my desktop computer, my friends etc ).

I really love this method, but currently, I can't boot on my Fedora, I only have the legacy bios, and not the UEFI. SeaBIOS list my Corsair device, but freeze when I try to boot it.   

There is any chance to support UEFI, and if yes, do you have any idea of when ? ( days, weeks, months ?)

MrChromebox, 2019-06-13:
under Settings / About ChromeOS / Detailed build info, what's the firmware branch/version listed?

lakano, 2019-06-13:
Thanks for you quick answer! 
Platform: 11895.118.0 (Official Build) stable-channel nami
Firmware: Google_Nami.10775.101.0 / stable
ARC: 5563251
V8: 7.4.288.28

MrChromebox, 2019-06-13:
hmm, I don't see that as a Nami variant, will have to investigate. I was planning to support Nami boards in the next release, but SONA isn't one I have support for (just VAYNE, PANTHEON, and AKALI).

lakano, 2019-06-13:
Ok, thanks for your answer. So, I suppose this will not be supporter before many weeks or months, right ?
So, I'll return my chromebook next week to prevent any charges.

MrChromebox, 2019-06-13:
you can try a test build if you want, but you'll need to have a debug cable to disable firmware write protect. shoot me an email @ gmail

MrChromebox, 2019-08-29:
Sona added as part of 2019.08.22 release

closed 2019-08-29 by MrChromebox