MrChromebox/firmware #149: Need help - Asus Chromebox CN60 - possibly bricked?

justforfun20, 2019-06-16:
Long time user setup with dual boot - ChromeOS + LibreElec.  I tried to upgrade LibreElec to 9.0.2 from 8.2.5 but couldn't due to lack of space.  I decided to reinstall to Standalone and that is where things went wrong.  Long story short, I tried late at night and screwed things up royally.  Now I can't load the setup-kodi.sh script anymore... get various errors everytime I try.  Most recent error is "Error downloading one or more required files; cannot continue" followed by "curl: (6) Couldn't resolve host 'mrchromebox.tech'.

Any help would be greatly appreciated.  Ideally, I just want LibreElec v9 as a standalone.  I don't use ChromeOS on this device at all.

justforfun20, 2019-06-16:
Just tried again... I am able to ping to www.google.com so I know my network is connected.  After confirming the ping, i tried again and now got a different error... 

Error downloading cbstool; cannot proceed.
Unable to download cbfstool utility; cannot proceed.

justforfun20, 2019-06-16:
Forgot to mention... when booting, if I don't press any keys, I get the following message/error:

SeaBIOS (version rel-1.9.0-126-g7dd4d62-20160418-MattDevo)

Booting from Hard Disk...
Booting...

Failed to load ldlinux.c32
Boot failed: please change disks and press a key to continue.

This is the original error which I then tried to log back into the script but no luck/success as per above.

MrChromebox, 2019-06-16:
ensure the date/time are set correctly so the SSL certs are valid. if
you're able to connect to wifi, may take a min to sync

justforfun20, 2019-06-16:
Thanks @MrChromebox .  I'm now stuck at the "Failed to load ldlinux.c32" screen.  How would I go about ensuring the date/time are set correctly?

justforfun20, 2019-06-16:
Another attempt... another error... after running the setup-kodi.sh

Error downloading flashrom: cannot proceed.
Unable to download flashrom utility: cannot continue

MrChromebox, 2019-06-16:
you're running the script from ChromeOS? there shouldn't be any issues doing so, I just tested it here. It's possible there's some communication issue with the server from your location, in which case retrying may help.  You can also just do a full reset via ChromeOS Recovery to ensure that everything is sane and your running an up-to-date version of ChromeOS, and then retry

justforfun20, 2019-06-16:
Some progress made... i moved the box closer to the router so that I could physically connect to network (previously connected over wifi).  I was able to load the script and run the dual boot chrome + libreelec option...

But then more errors:

LibreELEC / Dual Boot Install
Stage 1 / repartitioning completed, moving on.
Stage 2: Installing LibreELEC
Partitions formatted and mounted
Updating bootloader
Downloading LibreELEC
LibreELEC download complete; installing...

cp: error writing '/tmp/System/SYSTEM'; No space left on device
LibreELEC Installation Complete

Press [Enter] to return to the main menu.

I pressed [Enter] ... system rebooted and started loading LibreELEC... but then error again

*** Error in prepare_sysroot: mount_common: Could not mount /dev/SYSTEM ***
### Starting debugging shell for boot step: prepare_sysroot... type exit to quit ###

After waiting for 3 minutes, I typed exit

# exit
mount: mounting /flash on /syssroot/flash failed: No such file or directory
mount: mounting /storage on /sysroot/storage failed: No such file or directory
*** Error in prepare_sysroot: final_check: Could not find systemd! ***
### Starting debugging shell for boot step: prepare_sysroot... type exit to quit ###

After waiting for 3 minutes, I typed exit

# exit
mount: mounting /dev on /sysroot/dev failed: No such file or directory
mount: mounting /proc on /sysroot/pro failed: No such file or directory
mount: mounting /sys on /sysroot/sys failed: No such file or directory
BusyBox v1.29.3 (2019-01-29 11:23:58 EST) multi-call binary.

Usage: switch_root [-c CONSOLE_DEV] NEW_ROOT NEW_INIT [ARGS]

... and I can go on...

Any ideas on how to troubleshoot/fix further?

MrChromebox, 2019-06-16:
why are you running the dual boot option, when you said you want a standalone setup?  And the script error clearly told you what the issue was when installing, so of course LE isn't going to work at this point.

You really need to do a full reset via Recovery Mode, and start with a clean slate, regardless of which option you want.

justforfun20, 2019-06-16:
Thanks @MrChromebox.  I'm going to do a full reset.  The Chromebook Recovery Utility is unfortunately not writing to the USB or SD card which I have tried.  Same result letting the tool download vs. using a local image (which I downloaded from https://cros-updates-serving.appspot.com/).  It just stays at "Writing: 0% completed"

Any suggestions?

MrChromebox, 2019-06-16:
try writing the unzipped .bin with Rufus or Win32DiskImager.  Stuck at 0% means the app couldn't get a lock on the drive to start writing for some reason.  Or try from another device if possible

justforfun20, 2019-06-17:
Thanks again for all you advice @MrChromebox .  I successfully completed the full reset via Recovery and installed LibreELEC 9.0.2 in standalone mode.

closed 2019-06-17 by MrChromebox