MrChromebox/firmware #150: Fan not running on Dell 7310 (LULU)

VincentGijsen, 2019-06-21:
dear MrChromeBox,

i've a Dell 7310 and am running the UEFI firmware (latest as of this writing), but notice the fan would never switch on, and runs rather hot

I read a closed issue (https://github.com/MrChromebox/firmware/issues/115) that the command 
```
chronos@localhost /home/user/86...c/Downloads $ sudo ectool pwmgetfanrpm
EC returned error result code 1
Fan 0 RPM: 0
``` 

not sure what to expect here, but the error doesn't sound great?

I tried this command from a Cloudready install, but noticed the same behavior on my ParrotOS install

```
Linux localhost 4.14.105+ #15 SMP Tue May 21 01:11:10 UTC 2019 x86_64 Intel(R) Celeron(R) CPU 3215U @ 1.70GHz GenuineIntel GNU/Linux
```
anything you can suggest to try to pinpoint the issue?

tnx

MrChromebox, 2019-06-21:
the fan should be put in automatic mode upon a hard reset (Refresh+Power), but you can always set it manually using pwm*set*fanrpm instead of get. Start in the 2000-3000 range, higher if you want something clearly audible for validation

VincentGijsen, 2019-06-21:
i'll  try the hard-reset mode, however forcing it manually doesn't seem to work:
```
chronos@localhost /home/user/8....b3c/Downloads $ sudo ectool fanduty 10
EC returned error result code 6
EC returned error result code 1
Fan duty cycle set for all fans.

``

MrChromebox, 2019-08-07:
the 4.10 release will (re)set the fan control to auto mode on each boot, should resolve this

closed 2019-08-07 by MrChromebox

josephasmith1, 2020-03-23:
Mr Chromebox- Dell 7310 Lulu i3 user, recently updated to latest firmware and Windows 10 now runs the fan much harder than before. I never had an issue with the fan not running or the processor getting hot in previous releases. Now it seems to blow full speed 50-75% of the time. Also it seems to have done something to the modulation of the fan making sounds I never heard this laptop make (like reving up and down quickly). Possible the i3 is different than the base celeron? Everything else is perfect.

MrChromebox, 2020-03-23:
@josephasmith1 I did adjust the fan on Lulu to come on earlier at lower temps, but the RPMs are lower too. the upper RPMs / fan speeds shouldn't be changed

josephasmith1, 2020-03-23:
Hey thanks for the reply. Definitely something weird with the way the fan trails off. It's almost like a burbling sound. You can hear it at the end of this audio clip. The fan is also speeding up a lot now, when even before in ChromeOS it was always really quiet. Not sure if the fan speeds/temps were turned against the celeron or i3, but just a thought. https://drive.google.com/open?id=1arZStsEtGmLCJaMLrGjn_ltieVzmCZSO

MrChromebox, 2020-03-24:
@josephasmith1 shoot me an email mrchromebox at gmail and I'll provide a link for you to test. I'm not able to reproduce here on my Celeron LULU