MrChromebox/firmware #153: IOMMU support

shuxin, 2019-07-02:
my asus CN60 has an i7-4600U supported vt-d and iommu. but when I add intel_iommu=on in grub. system hang up when start.

MrChromebox, 2019-07-02:
distro/kernel/version? firmware version?

bobreidx, 2021-03-20:
I have a celeron CN60 running proxmox, and am considering an i7 CN60. Are there specifics as to how to ensure vt-d to run under proxmox with a i7-4600U?
Do you have guidance on what distros or firmwares that would allow Intel's IOMMU to function on a CN60 if not proxmox?
THANKS!

MrChromebox, 2021-03-20:
generally speaking, IOMMU isn't enabled by default in most kernels, and you need to add kernel param `intel_iommu=on` to enable it. The stock firmware definitely won't cut it, so my UEFI firmware will be the only option. I've not tested IOMMU on Haswell in some time but it should work. If not, I can likely fix it easily enough since it's enabled/tested for newer platforms including Broadwell

MrChromebox, 2025-06-27:
no idea why Proxmox would report that, IOMMU is enabled in the firmware. Haven't had time to investigate