MrChromebox/firmware #154: ASUS Chromebox CN60 No Boot

ahmadsp, 2019-07-17:
Just flashed the UEFI ROM firmware on an ASUS Chromebox CN60 on Gallium OS, 

Rebooted, and now nothing is responding. Monitor is asleep, power turns on and off device, no response whatsoever. Is this bricked?

MrChromebox, 2019-07-18:
what firmware were you running before? stock + RW_LEGACY, or an older SeaBIOS Full Rom? I've seem some issues with the latter, but unable to pinpoint the issue.

Also, what display and connection type are you using?

ahmadsp, 2019-07-18:
I was running the SeaBIOS ROM by John Lewis.

I'm using an HDMI to DisplayPort adapter on an Acer K272HL monitor

MrChromebox, 2019-07-18:
are you able to try without the adapter? the box does have native DisplayPort out

ahmadsp, 2019-07-18:
I just plugged in an HDMI cable since this box and the monitor both support HDMI and now I'm seeing something that says "Graphics Console Started"

MrChromebox, 2019-07-18:
that typically happens when you plug in the monitor after boot, and it units without an attached display

ahmadsp, 2019-07-18:
Ok, I rebooted, and now it's showing the UEFI interactive shell, the mapping table, and it mentions to press ESC in 1 sec to skip startup.sh, then it shows Shell>, but no boot. Any ideas?

MrChromebox, 2019-07-18:
now you install a UEFI bootable OS from usb. press ECS on boot, or type exit / press enter from shell, then go to Boot Manager and select the usb to boot

ahmadsp, 2019-07-18:
Hey, it’s working now! Wasn’t too complex of an issue but you’ve been a
great help! Thanks!

*From:* MrChromebox <notifications@github.com>
*Sent:* Thursday, July 18, 2019 8:23 AM
*To:* MrChromebox/firmware <firmware@noreply.github.com>
*Cc:* ahmadsp <asepehrnia@americor.com>; Author <author@noreply.github.com>
*Subject:* Re: [MrChromebox/firmware] ASUS Chromebox CN60 No Boot (#154)

now you install a UEFI bootable OS from usb. press ECS on boot, or type
exit / press enter from shell, then go to Boot Manager and select the usb
to boot

—

*This email may be
confidential or privileged. If you received this 
communication by mistake,
please don't forward it to anyone else, please 
erase all copies and
attachments, and please let me know that it went to 
the wrong person. Thanks.
The above terms reflect a potential business 
arrangement, are provided solely
as a basis for further discussion, and are 
not intended to be and do not
constitute a legally binding obligation. No 
legally binding obligations will be
created, implied, or inferred until an 
agreement in final form is executed in
writing by all parties involved*

closed 2019-07-18 by MrChromebox