MrChromebox/firmware #155: Gemini Lake, CR5, Grabbiter

robertmaynord, 2019-07-29:
Our school was using DELL 3189 Braswell machines, and they worked fine with Mrchromebox firmware. However, Dell has discontinued the 3189, and replaced it with the 3100 series.  On the 3100, I am attempting to use the Mrchromebox firmware utility, which shows RW_Legacy firmware (but not Full Rom).  When I select "1) Install RW-Legacy" nothing happens - the screen just stays the same.  

The Dell 3100 uses the **Gemini Lake** chipset and the **Grabbiter** base.  I know that Grabbiter is not listed as a supported Mrchromebook device, however I thought it might at least work with RW_Legacy.  I am now becoming suspicious that the problem might be related to the CR5 security chip that our Braswell 3189 did not have (it had a motherboard screw). 

Is there any one who has knowledge of the Gemini Lake - Grabbiter combo?  Should I be able to make it work with RW_Legacy?  Or is there a comparable new chromebook that is still being manufactured that will work for my students (durable case, 4MG ram, 32MG storage, touchscreen)?

I appreciate any thoughts - the students are coming soon!

MrChromebox, 2019-07-29:
none of the Gemini Lake devices are supported yet, so if the script isn't reflecting that then I need to fix.  RW_LEGACY seems to be broken on Google's end on a lot of current-model devices right now so hard to recommend anything

closed 2022-04-09 by MrChromebox

frankk74, 2023-04-15:
As of April 2023 it all seems to work! Thank MrChromeBox you turned a useless chromebook into a real Linux Laptop!!!!