MrChromebox/firmware #156: accidental lock ups

Baker-1, 2019-08-15:
Hi,

At 5th of march I installed the BIOS MrChromebox-4.9 01/04/2019 on my Dell Chromebook 13 (7310) (LULU). Since then the laptop with linux OpenSuse sometimes locked up. The kernel parameter 'intel_idle.max_cstate=2' solved this issiue, but also dropped the battery life time. This kernel parameter prevents using CPU C3 sleeping states and lower. The lock ups were caused by using these battery saving processor states.
Since there is no recent Linux kernel issue on Intel Broadwell processors with the kernel S3 sleeping state and it started when installing the MrChromebox-4.9 BIOS, I assume the cause is the BIOS.

Could this be BIOS related and is there a way to solve this?

Information about the machine:
Hardware name: GOOGLE Lulu/Lulu, BIOS MrChromebox-4.9 01/04/2019
Device: Dell Chromebook 13 (7310) (LULU)
CPU Type: Intel Broadwell
Kernel: 5.1.16

I must also make a compliment.
The installation of the MrChromebox-4.9 BIOS did solve the issues i had with hibernation and the i915 display driver. After installing the new MrChromebox BIOS, hibernation worked and the display driver worked without extra options or kernel adjustments. Thanks for that!

MrChromebox, 2019-08-15:
I would try testing against an earlier kernel (4.1x) and see if the issue persists, just to ensure it's not a kernel regression (possibly one that only affects devices running coreboot). I'm not aware of any general issues with coreboot/Broadwell/kernel 5.x but probably not a lot of data points yet

Baker-1, 2019-08-19:
And yesterday I suddenly had a hard lock up again with the 5.1.16 
kernel. Testing by just leaving the laptop on with a light task is a 
good test I think.

I will no try the 4.14.139 kernel for some time.

closed 2026-11-04 by MrChromebox