MrChromebox/firmware #160: No sound on Lenovo N42

Mutexistante, 2019-09-15:
Hi everyone.After getting rid of ChromeOS my laptop has no sound from speakers, and (previously did't note) no built-in microphone detected. I`ve posted this issue on every help forum out there, but it seems nobody knows how to fix it,tried linux mint 19.2 and GalliumOS so i guess it could possibly be a firmware problem (UEFI full ROM)

# dmidecode 3.1
Getting SMBIOS data from sysfs.
SMBIOS 2.8 present.

Handle 0x0002, DMI type 2, 14 bytes
Base Board Information
	Manufacturer: GOOGLE
	Product Name: Reks
	Version: 1.0
	Serial Number: 123456789
	Asset Tag: Not Specified
	Features: None
	Location In Chassis: Not Specified
	Chassis Handle: 0x0003
	Type: Unknown

Handle 0x000F, DMI type 41, 11 bytes
Onboard Device
	Reference Designation: VGA compatible controller
	Type: Video
	Status: Enabled
	Type Instance: 0
	Bus Address: 0000:00:02.0

Handle 0x0010, DMI type 41, 11 bytes
Onboard Device
	Reference Designation: ???
	Type: Sound
	Status: Enabled
	Type Instance: 0
	Bus Address: 0000:00:15.0

Thanks in advance.

MrChromebox, 2019-09-15:
audio on REKS and other Braswell devices is fully working, just needs OS support. Works under GalliumOS 3.0, Manjaro 18.1+, Windows, and others. Nothing to do with firmware I'm afraid

closed 2019-09-15 by MrChromebox

Mutexistante, 2019-09-16:
Thank you MrChromebox, it seems more likely that it could be fixed sooner on Gallium than other OS`s.

MrChromebox, 2019-09-16:
it was fixed well over a year ago, maybe two. Audio is fully working under GalliumOS 3.0 on Braswell devices, even booted in live mode. Be sure you're using the Braswell ISO

moriel5, 2019-09-16:
Have you tried Solus?
I find that support is usually on par with, or even better than, Gallium OS for the older Chromebooks, simply due to it being a rolling release (which manages to retain it's stability), which also means that the kernel is usually pretty much up to date (though one major version behind, to retain stability).
Solus is a curated distro though, so the software collection is smaller (though it is growing).

MrChromebox, 2019-09-16:
not recently. When I did, it worked fine on Haswell/Broadwell devices, but not on Baytrail/Braswell.  Atom-based platforms have traditionally struggled to get mainline kernel support, though that should be improving now that there's one driver for both (previously there were two incompatible ones, so you had to choose at kernel compile time)

moriel5, 2019-09-16:
Yeah, I remember the audio issues on Baytrail (as well as the keyboard, touchpad, and usb issues, amongst others).
That was around the time I had flashed Coreboot on my sister's laptop.
So for a week or so it ran Gallium OS, with an ancient (heavily patched) kernel, until most of the issues were fixed on the mainline kernel, at which point I switched it over to Solus.

Do you remember my bug report about audio not working? Since it was the only thing unfixed back at the time, and until it was fixed (and stable), we used a USB audio card, as well as a bluetooth speaker (there were bluetooth issues occasionally as well, despite the card being an Intel Wireless-AC 7260, since upgraded to the 9260).