MrChromebox/firmware #161: Possible support for using FWUpd via FlashROM?

moriel5, 2019-09-16:
What do you think about the possibility of using something like the upcoming GNOME Firmware Manager to update your firmwares?

Perhaps it would be a good idea to be able to use FWUpd to update the firmware, with FlashROM doing the actual work?

MrChromebox, 2019-09-16:
fwupd doen't support ChromeOS flashrom, which is what I use. I'd have to see about adapting to using upstream flashrom instead. Not a high priority, but may come out of work I'm doing for fwupd integration for Purism laptops -- so we'll see

moriel5, 2019-09-16:
Thanks, that would be great.
Can I help with getting the Chromebooks/Chromeboxes approved by upstream FlashROM?
Since otherwise, FlashROM doesn't allow flashing anything via internal means.

I'm also the (currently unofficial) maintainer of the FlashROM package on Solus, so at least there I could get FlashROM to be updated quickly (Solus is a rolling release distro, and we sync the stable branch once a week).

MrChromebox, 2019-09-16:
> Can I help with getting the Chromebooks/Chromeboxes approved by upstream FlashROM?

there's nothing to approve, they work fine, it's just a matter of upstream flashrom not supporting all of the features that ChromeOS flashrom does

> Since otherwise, FlashROM doesn't allow flashing anything via internal means.

not sure what you mean here

moriel5, 2019-09-16:
Upstream FlashROM appears to have a whitelist for laptops, since if it is run on laptops that are not officially supported, it fails to run the internal programmer, and puts up a warning about the laptop not being supported, with a link to the FlashROM wiki.

MrChromebox, 2019-09-16:
which is easily circumvented with '-p internal:laptop=force_I_want_a_brick' so not sure that's a real issue. Whitelisting ChromeOS devices would be useful to avoid that, but not an actual issue right now.

The main issues are upstream flashrom's lack of support for FMAP regions, and lack of support for en/disabling software write-protect and setting the protected region.

moriel5, 2019-09-16:
Hmm... Is there any progress on those in mainline?

MrChromebox, 2019-09-16:
I'm not sure, haven't checked lately

moriel5, 2019-09-16:
I'll try looking into it. I can't contribute code, since I still haven't learned enough to be able to code anything beyond the most basic of HTML5, but at least I can try getting a grasp of the situation.

MrChromebox, 2019-09-16:
talk to the gents in #flashrom on IRC

moriel5, 2019-09-16:
Thanks, I'll do so when I have the time.

moriel5, 2019-09-16:
I haven't had time to speak on the #flashrom IRC channel yet, however I took a quick look at the commit logs, bugtracker, and mailing lists (via the handy search option, of course), and it looks like FMAP support has been in mainline for some time already, along with disabling write-protect (not sure about enabling, however that may also be supported). No idea about support for setting the protected region yet.

closed 2026-11-04 by MrChromebox