MrChromebox/firmware #162: Adding a nicer GUI, perhaps via Project MU

moriel5, 2019-09-16:
This is very low priority, however it would be nice to have a GUI that is not necessarily text-based, with support for additional inputs other than the keyboard.
A possible candidate may be Microsoft's Project MU's GUI.

I personally am perfectly fine with the current GUI, however many people would rather have something else, and now that we have Chromebook tablets, it would be a good idea to add support for mouse and touch inputs.

MrChromebox, 2019-09-16:
the CorebootPayloadPkg Tianocore component I'm using has been deprecated/removed from upstream, so I'm not going to spend any more time on it. Once UefiPayloadPkg matures a bit I'll see what can be done to spruce it up, but very, very low priority

moriel5, 2019-09-16:
No problem, I had only created this issue to bring up a suggestion, Microsoft's Project MU is only one of many options.
Should I close this issue now?

moriel5, 2020-08-02:
Since CorebootPayloadPkg has been replaced by UefiPayloadPkg, I am bringing attention to this issue.

dragon788, 2020-08-19:
Does rEFInd work alongside Tianocore?

MrChromebox, 2020-08-19:
> Does rEFInd work alongside Tianocore?

yep, been using it for years

moriel5, 2020-09-02:
Issue #215 got me thinking that as part of a nicer GUI for for the firmware settings, showing the battery state would be great.

MrChromebox, 2020-09-02:
it would be, but either need to directly interface with the EC, or extract from ACPI tables

closed 2026-11-04 by MrChromebox