MrChromebox/firmware #165: Some or most Linux distros do not boot in full UEFI firmware version 4.10 (on AKALI360, WUKONG and presumably other boards as well)

FantasyCookie17, 2019-10-16:
This includes, but probably is not limited to GalliumOS, Sparkylinux, Void Linux and NixOS. They show a black screen, the firmware doesn't even recognize they aren't booting properly. Here are the links to at least some of the Reddit posts pointing this issue out: 
https://www.reddit.com/r/GalliumOS/comments/dib0ia/most_linux_distros_including_gallium_will_not/ 
https://www.reddit.com/r/GalliumOS/comments/disxzv/wukong_kabylake_chromebox_issues/
MrChromebox already suggests in the latter one it might be due to the GRUB config on the non-working distros.

MrChromebox, 2019-10-17:
looking into this. It's definite an issue with the grub EFI stub on those ISOs, but the question remains on what changed between the 4.9 and 4.10 releases which caused things to break

FantasyCookie17, 2019-10-23:
I also tried out the Gentoo minimal installer and OpenBSD install system out now. The former has the same issue with black screen, the latter stops continuing to boot at some point - or the screen is just kind of messed up in OpenBSD (it seems to start the shell in the background, however just a chaos of dots at the upper border, while the boot screen is still visible, not had that kind of issue on my desktop PC) Oh, and Arch works as well, but this was kind of to be expected because Manjaro worked.

FantasyCookie17, 2019-10-25:
@MrChromebox Is there anything I could do to help with finding the issue? I’m able to do a bit of C programming (however I haven’t worked with bit operators and memory addresses yet, although I could learn to do so), so if there is the possibility I could help you there, just tell me where to look (or something like that). I’m always available for testing distributions or new releases as well (unless they will brick my device; my SPI programmer I ordered for disabling the ME won’t arrive before next month). 
If you/we can’t fix it soon, I guess I’ll just go with the hacky “solution” you suggested in /u/smtylr’s thread.

MrChromebox, 2019-10-25:
I've figured out thr issue, will post fixed builds this weekend

FantasyCookie17, 2019-10-25:
Okay, great. I’ll probably contribute to writing your firmware images anyway at some point when I have enough time and there’s something I can help with.

MrChromebox, 2019-10-28:
fixed in 2019.10.28 release

closed 2019-10-28 by MrChromebox