MrChromebox/firmware #167: Firmware request

VincentRegios, 2019-10-21:
I'm looking for a firmware for acer chromebook spin 15 cp315-1h-p64s the one with an intel pentium N4200 processor. /u/MrChromebox.

MrChromebox, 2019-10-21:
no plans for UEFI/Full ROM firmware for Apollolake devices. Come to #coreboot on IRC after reading docs :)

RodBarnes, 2022-04-12:
I think this issue is now closed?  At least, when I run the firmware script on on my Acer CB515-1HT Apollo Lake, the Legacy firmware is now grayed out and the UEFI firmware is now enabled.

MrChromebox, 2022-04-12:
@RodBarnes RW_LEGACY shouldn't be grayed out for APL

closed 2022-04-12 by MrChromebox

RodBarnes, 2022-04-12:
[Screenshot](https://i.postimg.cc/c4CJryjr/Screenshot-2022-04-12-3-20-24-PM.png)

MrChromebox, 2022-04-13:
ugh, ok, will see what I broke then

RodBarnes, 2022-04-18:
Just circling back to confirm this is fixed.  I went through the steps to download and run the script and it now shows Legacy firmware is, again, available.