MrChromebox/firmware #168: Power LED on Dell 3181 not working

gruberaaron, 2019-10-24:
Power LED on lid of Dell Inspiron 11 3181 not working

FantasyCookie17, 2019-10-25:
This may not even be a firmware issue - you could e.g. try to install and use the light or brightnessctl package of your distribution and see if those can change the status of the LED.

gruberaaron, 2019-10-31:
$ brightnessctl -l
Available devices:
Device 'intel_backlight' of class 'backlight':
Current brightness: 472 (50%)
Max brightness: 937

Device 'phy0-led' of class 'leds':
Current brightness: 1 (100%)
Max brightness: 1

Device 'input2::numlock' of class 'leds':
Current brightness: 0 (0%)
Max brightness: 1

Device 'input2::capslock' of class 'leds':
Current brightness: 0 (0%)
Max brightness: 1

Device 'input2::scrolllock' of class 'leds':
Current brightness: 0 (0%)
Max brightness: 1

Device 'mmc1::' of class 'leds':
Current brightness: 0 (0%)
Max brightness: 255

Device 'mmc0::' of class 'leds':
Current brightness: 0 (0%)
Max brightness: 255

Tried mmc0 and mmc1 to 100% but did not enable the power LED.

I am not sure what the 'light' package is as I cannot find it.

MrChromebox, 2019-10-31:
so is this a secondary power LED?  The LEDs are all controlled by the EC, so unless my EC firmware borked something, it seems odd that it would stop working. This is a KEFKA device?

FantasyCookie17, 2019-10-31:
I am not sure what the 'light' package is as I cannot find it.
> […](#)
light is essentially another CLI program with similar functionality to brightnessctl. Some distros, e.g. Arch, include both in their repos, some, e.g. Gentoo, only have light, and some, apparently what you are using, only include brightnessctl.

FantasyCookie17, 2019-10-31:
> I am not sure what the 'light' package is as I cannot find it.
> […](#)

light is essentially another CLI program with similar functionality to brightnessctl. Some distros, e.g. Arch, include both in their repos, some, e.g. Gentoo, only have light, and some, apparently what you are using, only include brightnessctl.

closed 2025-10-23 by gruberaaron