MrChromebox/firmware #170: Thermal Shutdowns Teemo / Kench - Fan speed not increasing enough

stevefrancesco, 2019-11-09:
I have two chrome boxes with identical configurations both flashed with Full Roms a couple of days ago i7-8650U Asus / HP (identical MoBo's). 

However in Windows 10 they appear to thermally shutdown during any stressful task e.g. benchmarking. The same result in games which will cause windows to shutdown most of the time before even loading. 

The fans, while they do increase in speed a bit under load they do not get near to their potential maximum speed (Which you can hear when the units initially start up at full fan speed) - they reach max temp very fast, within seconds. 

The two boxes work beautifully in OSX - actually beating the i3 2018 Mac Mini substantially in Geekbench (awesome!) - but I do not believe that OSX has the same thermal safeguards as windows which is a little scary that these units "may" burn out prematurely under regular heavy loads unless fan speeds cans be increased by a good margin.

MrChromebox, 2019-11-09:
the fan speed is controlled by the EC and is purely a function of temperature, unless overridden by the OS. I've not spent any time validating Windows on FIZZ boxes, so YMMV. It's not exactly easy to debug these sorts of things, and Windows is not something I'm particularly inclined to spend any time on.

stevefrancesco, 2019-11-09:
Fans are not even seen by the OS, hence the reason I thought they might be a function of the firmware - oh well :( 

Great Job anyhoo ;) 

On another note, I cant seem to get the Bootloader to load an OS without manually selecting from maintenance manager every time 

diskname/EFI/BOOT/BOOTX64.efi. Something I'm missing ?

MrChromebox, 2019-11-09:
the main/AP firmware doesn't provide any interface to see/control them (though it does set the fan mode to auto at boot) - the EC handles it. Under Linux, ectool can be built/used to modify fan settings. No idea how one might do that under Windows.

for the boot issue, sounds like the bootorder variable got corrupted. Boot a Linux USB, run my script, use the clear NVRAM option, reboot.

stevefrancesco, 2019-11-09:
Will give it a shot - thanks again

stevefrancesco, 2019-11-10:
FYI the NVRAM reset worked a treat

closed 2019-11-10 by stevefrancesco