MrChromebox/firmware #171: hp13g1 cpu frequency locked

Sp1riTung, 2019-11-16:
CB: hp cb13g1

cpu:m5-6y57

power charger:45w pd charger type-c

system: win10 1809 x64

bios:mrchromebox 4.1- coreboot

hi, I found that my laptop always runs at 1.1ghz at Windows10 while the basic frequency seems should be at 1.51ghz according to the System Explorer. :(

I guess that maybe it is the result of the bios default setting??. Thanks for advice.

Sp1riTung, 2019-11-16:
it just cannot Turbo automatically and always runs at 1.1ghz or lower

MrChromebox, 2019-11-16:
there are many reasons Skylake devices are completely unsupported for running Windows, and power management is one of those.  If the problem exists under Linux I'll investigate, but otherwise I'm afraid that's just how it is

Sp1riTung, 2019-11-16:
> there are many reasons Skylake devices are completely unsupported for running Windows, and power management is one of those. If the problem exists under Linux I'll investigate, but otherwise I'm afraid that's just how it is

Thanks a lot. Maybe I will try Linux later. I will reply then.
btw, considering that I have already installed Windows and coreboot, is it just OK to act in the normal way to install Linux( like Ubuntu or Gallium OS,etc.) under windows system? like using flashdisk to boot and follow the ordinary online guides.

Gratitude for your contribution!

MrChromebox, 2019-11-16:
>  is it just OK to act in the normal way to install Linux( like Ubuntu or Gallium OS,etc.) under windows system?

you can just live boot a Linux ISO, no need to install anything. Since you flash the firmware just works like a normal PC laptop in that regard

closed 2026-11-04 by MrChromebox