MrChromebox/firmware #172: Hp 11 Chromebook touchscreen not working

pcs3rd, 2019-11-17:
The touchscreen varient of hp 11-v025 doesn't identify the touchscreen on any Linux is while using custom coreboot uefi firmware, does work on legacy firmware though

MrChromebox, 2019-11-17:
please provide output of dmesg and `cat /proc/interrupts` from both legacy and UEFI firmware so I can see what's going on

pcs3rd, 2019-11-18:
UEFI outputs this  
Will get to bios tomorrow hopefully  

```
          CPU0       CPU1       
  0:          7          0   IO-APIC   2-edge      timer
  8:          1          0   IO-APIC   8-edge      rtc0
  9:          0       2501   IO-APIC   9-fasteoi   acpi
 18:      10327          0   IO-APIC  18-fasteoi   mmc1
 19:          0      91987   IO-APIC  19-fasteoi   mmc0
 29:        259          0   IO-APIC  29-fasteoi   intel_sst_driver
 32:       1028          0   IO-APIC  32-fasteoi   808622C1:00
 36:          0        906   IO-APIC  36-fasteoi   808622C1:04
 37:      82539          0   IO-APIC  37-fasteoi   808622C1:05
 42:          0          0   IO-APIC  42-fasteoi   dw:dmac-1
 43:          0          0   IO-APIC  43-fasteoi   dw:dmac-1
168:          1          0  chv-gpio  95  rt5645
182:          0        181  chv-gpio  17  i8042
183:          0       2562  chv-gpio  18  ELAN0000:00
184:          0          1  chv-gpio  19  SYTS7508:00
230:          0          0  chv-gpio  15  ACPI:Event
310:          0          0   PCI-MSI 458752-edge      PCIe PME, pciehp
311:          0          0   PCI-MSI 462848-edge      PCIe PME
312:      29171        716   PCI-MSI 327680-edge      xhci_hcd
313:          0     116988   PCI-MSI 1048576-edge      iwlwifi
314:      79872       1118   PCI-MSI 32768-edge      i915
315:        890          0   PCI-MSI 442368-edge      snd_hda_intel:card0
NMI:         31         31   Non-maskable interrupts
LOC:     143486     162332   Local timer interrupts
SPU:          0          0   Spurious interrupts
PMI:         31         31   Performance monitoring interrupts
IWI:       6167          1   IRQ work interrupts
RTR:          0          0   APIC ICR read retries
RES:     161761     160026   Rescheduling interrupts
CAL:      16628       8489   Function call interrupts
TLB:       9279       8676   TLB shootdowns
TRM:          0          0   Thermal event interrupts
THR:          0          0   Threshold APIC interrupts
DFR:          0          0   Deferred Error APIC interrupts
MCE:          0          0   Machine check exceptions
MCP:          2          2   Machine check polls
HYP:          0          0   Hypervisor callback interrupts
HRE:          0          0   Hyper-V reenlightenment interrupts
HVS:          0          0   Hyper-V stimer0 interrupts
ERR:          0
MIS:          0
PIN:          0          0   Posted-interrupt notification event
NPI:          0          0   Nested posted-interrupt event
PIW:          0          0   Posted-interrupt wakeup event

```

MrChromebox, 2019-11-18:
just pastebin or attach files please, too hard to compare otherwise

pcs3rd, 2019-11-18:
Oops, sorry, if you would like, I can take the previous comment and put it in a paste, bet here is the cat output from BIOS firmware pastebin.com/RdypYSRq

pcs3rd, 2019-11-26:
@MrChromebox poke? do you need anything else?

MrChromebox, 2019-11-26:
ah, I didn't get a notification that you had edited your post with the logs. I'll try to take a look over the next few days

MrChromebox, 2019-11-26:
quick scan of those logs shows both the stock firmware and UEFI firmware identifying and initializing the touchscreen identically. Everything appears to be working as it should from those logs, so I'm a bit baffled

pcs3rd, 2019-11-27:
Could it be an issue with Ubuntu?

MrChromebox, 2019-11-27:
possibly? I'd try live booting Manjaro, it has excellent Braswell support OOTB

pcs3rd, 2019-11-28:
ok.

pcs3rd, 2019-11-28:
@MrChromebox seems to be a conflict between the following kernel modules:  
elan_i2c  
i2c_hid  
so no issue on your end, probably issue with ubuntu itself and how they have the kernal configured

pcs3rd, 2019-11-28:
https://www.reddit.com/r/GalliumOS/comments/aien05/touchscreen_not_working_braswell_setzer/  
seems to be this exact issue :(

pcs3rd, 2019-12-19:
@MrChromebox it appears that monjaro works the opposite way. the touchscreen works without issue and now its the touchpad not working?

MrChromebox, 2019-12-19:
I'm not sure who that question is directed to, I don't have a Braswell device with a touchscreen to test. The devices I do have work without issue in every distro with a 4.14+ kernel

pcs3rd, 2019-12-20:
@MrChromebox I was asking in case maintainers broke something upstream. Switched to manjaro and I am currently having issues with random unresponsiveness issues with the touchpad. I am currently using the maintainer supplied libinput and elan_i2c kernal driver.
For example, this is the debug output from libinput.  
The DEVICE_ADDED tag is me modprobing the kernal driver.
```
event3   DEVICE_ADDED     Elan Touchpad                     seat0 default group11 cap:pg  size 95x52mm tap(dl off) left scroll-nat scroll-2fg-edge click-buttonareas-clickfinger dwt-on
 event3   POINTER_MOTION   +425.64s      -7.85/  1.81 (-26.00/ +6.00)
 event3   POINTER_MOTION   +425.71s     -26.50/  7.81 (-78.00/+23.00)
 event3   POINTER_MOTION   +426.04s     -107.35/ 32.61 (-316.00/+96.00)
 event3   POINTER_MOTION   +426.15s     -22.42/ -9.51 (-66.00/-28.00)
 event3   POINTER_MOTION   +426.37s     -49.94/-27.86 (-147.00/-82.00)
 event3   POINTER_MOTION   +426.73s     -29.89/-17.67 (-88.00/-52.00)
 event3   POINTER_MOTION   +428.17s       4.20/-15.90 (+14.00/-53.00)
 event3   POINTER_MOTION   +428.35s       4.70/-119.60 (+15.00/-382.00)
 event3   POINTER_MOTION   +428.49s      57.41/-38.05 (+169.00/-112.00)
 event3   POINTER_MOTION   +428.72s     124.67/ 40.09 (+367.00/+118.00)
 event3   POINTER_MOTION   +428.78s      27.52/ 20.38 (+81.00/+60.00)
 event3   POINTER_MOTION   +429.08s      91.04/ 73.38 (+268.00/+216.00)
 event3   POINTER_MOTION   +429.15s      20.38/ 18.68 (+60.00/+55.00)
 event3   POINTER_MOTION   +429.64s      92.40/108.37 (+272.00/+319.00)
 event3   POINTER_MOTION   +430.28s     -276.53/-125.35 (-814.00/-369.00)
 event3   POINTER_MOTION   +431.39s     225.23/ 92.74 (+663.00/+273.00)
 event3   POINTER_MOTION   +432.25s     -252.75/-68.28 (-744.00/-201.00)
 event3   POINTER_MOTION   +433.09s     -44.90/ -9.38 (-134.00/-28.00)
 event3   POINTER_MOTION   +433.11s      -5.36/ -3.69 (-16.00/-11.00)
 event3   POINTER_MOTION   +433.15s     -13.93/ -9.17 (-41.00/-27.00)
 event3   POINTER_MOTION   +433.18s      -9.85/ -5.78 (-29.00/-17.00)
 event3   POINTER_MOTION   +433.24s     -33.97/-31.93 (-100.00/-94.00)
 event3   POINTER_MOTION   +434.11s      56.73/ -3.74 (+167.00/-11.00)
 event3   POINTER_MOTION   +434.80s     296.23/ 66.58 (+872.00/+196.00)
 event3   POINTER_MOTION   +435.47s     -298.27/ 75.76 (-878.00/+223.00)
 event3   POINTER_MOTION   +436.38s     178.35/-92.74 (+525.00/-273.00)
 event3   POINTER_MOTION   +436.44s      36.01/  8.15 (+106.00/+24.00)
 event3   POINTER_MOTION   +437.36s     -363.49/-70.32 (-1070.00/-207.00)
 event3   POINTER_MOTION   +437.61s      -1.23/  1.85 ( -4.00/ +6.00)
 event3   POINTER_MOTION   +438.02s     318.73/ 49.27 (+1035.00/+160.00)
 event3   POINTER_MOTION   +438.28s     -279.92/-51.64 (-824.00/-152.00)
 event3   POINTER_MOTION   +438.92s       5.78/153.55 (+17.00/+452.00)
 event3   POINTER_MOTION   +438.94s     -17.33/ -7.13 (-51.00/-21.00)
 event3   POINTER_MOTION   +438.95s      -4.42/ -3.40 (-13.00/-10.00)
 event3   POINTER_MOTION   +438.99s     -15.63/-19.36 (-46.00/-57.00)
 event3   POINTER_MOTION   +439.04s      -9.51/-15.29 (-28.00/-45.00)
 event3   POINTER_MOTION   +439.17s      13.59/-70.32 (+40.00/-207.00)
 event3   POINTER_MOTION   +439.31s     103.27/-103.27 (+304.00/-304.00)
 event3   POINTER_MOTION   +439.46s     133.17/ -2.04 (+392.00/ -6.00)
 event3   POINTER_MOTION   +439.52s      44.16/ 12.23 (+130.00/+36.00)
 event3   POINTER_MOTION   +439.58s      21.40/ 10.19 (+63.00/+30.00)
 event3   POINTER_MOTION   +439.66s      28.88/ 16.31 (+85.00/+48.00)
 event3   POINTER_MOTION   +439.70s      21.74/  7.81 (+64.00/+23.00)
 event3   POINTER_MOTION   +440.04s     210.62/ -5.10 (+620.00/-15.00)
 event3   POINTER_MOTION   +440.93s      -1.68/  1.01 ( -5.00/ +3.00)
 event3   POINTER_MOTION   +440.96s     -15.12/  1.34 (-45.00/ +4.00)
 event3   POINTER_MOTION   +440.97s      -8.15/ -0.34 (-24.00/ -1.00)
 event3   POINTER_MOTION   +441.01s     -60.93/ -6.81 (-179.00/-20.00)
 event3   POINTER_MOTION   +441.06s     -52.42/ -5.79 (-154.00/-17.00)
 event3   POINTER_MOTION   +441.09s     -12.23/ -4.08 (-36.00/-12.00)
 event3   POINTER_MOTION   +441.11s     -10.19/ -3.06 (-30.00/ -9.00)
 event3   POINTER_MOTION   +441.12s     -10.19/ -3.06 (-30.00/ -9.00)
 event3   POINTER_MOTION   +441.13s      -5.10/ -2.38 (-15.00/ -7.00)
 event3   POINTER_MOTION   +441.16s     -16.65/ -5.78 (-49.00/-17.00)
 event3   POINTER_MOTION   +441.17s      -8.83/ -2.72 (-26.00/ -8.00)
 event3   POINTER_MOTION   +441.19s     -14.27/ -2.72 (-42.00/ -8.00)
 event3   POINTER_MOTION   +441.22s     -18.34/ -3.06 (-54.00/ -9.00)
 event3   POINTER_MOTION   +441.27s     -29.55/ -7.81 (-87.00/-23.00)
 event3   POINTER_MOTION   +441.29s     -10.53/ -3.06 (-31.00/ -9.00)
 event3   POINTER_MOTION   +441.36s     -51.98/-15.97 (-153.00/-47.00)
 event3   POINTER_MOTION   +441.39s     -23.44/ -2.72 (-69.00/ -8.00)
 event3   POINTER_MOTION   +441.40s      -6.45/ -2.04 (-19.00/ -6.00)
 event3   POINTER_MOTION   +441.42s     -10.53/  0.34 (-31.00/ +1.00)
 event3   POINTER_MOTION   +441.66s     368.14/177.85 (+1066.00/+515.00)
 event3   POINTER_MOTION   +441.87s     -240.80/-98.49 (-687.00/-281.00)
 event3   POINTER_MOTION   +442.56s     153.89/ 65.90 (+453.00/+194.00)
 event3   POINTER_MOTION   +442.92s     -244.93/-93.42 (-721.00/-275.00)
 event3   POINTER_MOTION   +442.95s     -17.67/-11.21 (-52.00/-33.00)
 event3   POINTER_MOTION   +442.96s      -3.74/ -2.72 (-11.00/ -8.00)
 event3   POINTER_MOTION   +443.04s     -36.01/-15.97 (-106.00/-47.00)
 event3   POINTER_MOTION   +443.06s      -5.78/ -3.06 (-17.00/ -9.00)
 event3   POINTER_MOTION   +443.10s     -16.31/ -8.15 (-48.00/-24.00)
 event3   POINTER_MOTION   +443.11s      -2.72/ -1.70 ( -8.00/ -5.00)
 event3   POINTER_MOTION   +443.12s      -1.70/ -1.36 ( -5.00/ -4.00)

```

closed 2019-12-20 by pcs3rd