MrChromebox/firmware #174: Boot manager has no items!

zzan1, 2019-11-20:
my device is Google chromebook pixel 2015 Samus. And now there is nothing in the bootmanager, that means i cant choose the boot options. Maybe i changed the order of BOOT i think because no i cant find this option. And i try Ubuntu, deepin and win10 installed by 'Boot maintenance->Boot from from file' and chose the boot Efi files. But there is noting changed and directly come into the BIOS. I think maybe the Boot manager is the reason. So i want to know hou can i do to fix it or any methods to reset the BIOS?

MrChromebox, 2019-11-20:
seems that Windows is corrupting the EFI bootloader variable. I'll have a fix in the next release, but in the meantime you can boot a Linux USB via Boot Maintenance Manager / Boot From File, then run my script and choose the option to Clear the NVRAM (c). That will fix things until Windows decides to break them again

zzan1, 2019-11-20:
Yeah, I tried it and was successful. And now i am trying to install ubantu. Thank you very much! Best wishes.

MrChromebox, 2019-11-26:
this issue is fixed in the 2019/11/25 release

closed 2019-11-26 by MrChromebox