MrChromebox/firmware #175: Firmware Request

shiveshram, 2019-11-23:
Hi!
I've got the new Pixelbook Go from Google and have been trying to install gallium os and update the firmware. However it is not recognizing my cpu or booting correctly. 

Thanks so much for all that you have done this project is amazing

MrChromebox, 2019-11-23:
while I'll likely add RW_LEGACY support for the PBG eventually, there's no guarantee it will work (it's broken on most recent devices), and even if it does, support in the mainline kernel is lacking and you'll certainly experience the same issues as other recent devices (no suspend/resume, no audio, touchscreen, etc).

In short, if you bought the PBG to run Linux natively, you should return it and buy a PC laptop.

Sapientone, 2019-12-17:
Hey - just wanted to say I think this is great. It would be awesome if you were able to get UEFI on PBG, but understand that it takes a lot of work. 

Thanks again

MrChromebox, 2019-12-17:
if you have a USB-C debug cable to disable WP, or are ok opening up the device and disconnecting the battery, I can provide you a test build -- shoot me an email at MrChromebox at gmail. The PBG is actually the only Kabylake device I haven't been able to test yet.

wez, 2020-01-08:
Just wanted to say that I've been successfully running @MrChromebox's test build on my PBG for the past couple of weeks.  I'm running Fedora 31 on mine and everything works, except for audio which has incomplete support in the 5.3 and 5.4 linux kernel.   I added `intel_idle.max_cstate=4 i915.enable_psr=0` to the kernel boot options to resolve a screen glitching/flashing effect that I had prior to switching away from chromeos and that got more prominent with a linux 5.3 kernel installed.  

I can also confirm that the even without the test build, the RW_LEGACY update that is currently available is sufficient to boot from USB so you can play around without going all-in, but note that suspending/resuming in that environment will require re-applying the legacy firmware update afterwards.

MrChromebox, 2020-01-08:
> but note that suspending/resuming in that environment will require re-applying the legacy firmware update afterwards.

actually, it's just the flag that gets cleared, so `sudo crossystem dev_boot_legacy=1` is all that's needed to restore :)

MrChromebox, 2020-06-07:
Pixelbook Go has been fully supported for some time now, closing

closed 2020-06-07 by MrChromebox